The-OpenROAD-Project/OpenLane #1599: FP_PDN_MACRO_HOOKS: No warning/error if instance does not exist

mole99, 2023-01-01:
### Description

Currently, no warning or error is issued if an instance specified in `FP_PDN_MACRO_HOOKS` does not exist.

For example, with the following entries for [caravel_user_project](https://github.com/efabless/caravel_user_project) the flow will complete successfully, even though `aaaa` does not exist as an instance.

```
"FP_PDN_MACRO_HOOKS": [
    "mprj vccd1 vssd1 vccd1 vssd1,",
    "aaaa vccd1 vssd1 vccd1 vssd1"
],
```

In this obvious case, the wrong entry can be easily spotted. But with subtle typos, it may take a while to find the actual problem.

### Proposal

Check if the modules specified in `FP_PDN_MACRO_HOOKS` actually exist. If not, issue a warning or an error.

d-m-bailey, 2023-01-01:
@mole99 Your `config.json` file has `"FP_PDN_CHECK_NODES": 0,`.
Doesn't that mean ignore any errors related to power domain connections?

Your final powered netlist contains instances named `\memory[0].sram` but your config file is missing the leading `\`
```
    "FP_PDN_MACRO_HOOKS": [
        "memory[0].sram vccd1 vssd1 vccd1 vssd1,",
        "memory[1].sram vccd1 vssd1 vccd1 vssd1,",
        "memory[2].sram vccd1 vssd1 vccd1 vssd1,",
        "memory[3].sram vccd1 vssd1 vccd1 vssd1,",
        "sram vccd1 vssd1 vccd1 vssd1"
    ],
```

Also `[` and `]` can be tricky, too (they may need to be escaped). There's a translated version of the config.tcl file in `runs/tag` where you can check.

mole99, 2023-01-01:
Hi @d-m-bailey, thanks for your reply!

I think this answer was meant for #1600, I will answer there.

But related to this issue, I set `"FP_PDN_CHECK_NODES": 1`, which did not generate any warning or error despite the wrong entry in `"FP_PDN_MACRO_HOOKS"`.

closed 2023-04-05 by donn