The-OpenROAD-Project/OpenLane #1603: Segfault in router (?)

xobs, 2023-01-03:
### Description

When running openlane to harden a design using caravel and sky130A, Step 7 (`Running Global Placement`) segfaults with a SIGABRT.

### Expected Behavior

Openroad should not segfault, and should either print an error indicating the issue, or continue without error.

### Environment report

```shell
Kernel: Linux v5.15.79.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 3af133706e554a740cfe60f21e773d9eaa41838c)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

Note: compressing using tar.gz or zip produces a 26 MB file, which I cannot upload. The attached file is a .tar.xz file that was renamed because the extension .xz is not allowed.

[issue_reproducible.tar.xz](https://github.com/The-OpenROAD-Project/OpenLane/files/10336625/issue_reproducible.tar.gz)
[7-global.log](https://github.com/The-OpenROAD-Project/OpenLane/files/10336639/7-global.log)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl
[ERROR]: Log: ../si/work/caravel-minimax/openlane/mimi/runs/23_01_03_20_01/logs/placement/7-global.log
[ERROR]: Last 10 lines:
38# 0x00007FF5A5C0EF1E in /lib64/libtcl8.5.so
39# Tcl_EvalEx in /lib64/libtcl8.5.so
40# Tcl_Eval in /lib64/libtcl8.5.so
41# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
42# ord::tclAppInit(Tcl_Interp*) in openroad
43# Tcl_Main in /lib64/libtcl8.5.so
44# main in openroad
45# __libc_start_main in /lib64/libc.so.6
46# 0x0000000000CAB507 in openroad
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
```

vijayank88, 2023-01-03:
@xobs 
Unable to extract the files. can you please upload again with `.zip` format?

vijayank88, 2023-01-03:
@xobs 
Have you tried with latest OpenLane tag?

xobs, 2023-01-03:
> @xobs Unable to extract the files. can you please upload again with `.zip` format?

I can't upload with `.zip` format because the filesize limit is 25 MB, and the file is 26 MB evein with `-9`.

I'm using the very latest `caravel_user_project`, and I just ran `make setup` today. Is there anything else you would like me to do?

vijayank88, 2023-01-04:
@xobs 
in [Makefile](https://github.com/efabless/caravel_user_project/blob/4fabf74c10ed2c9b72a15bb9fa5c756a5e4831dc/Makefile#L36) update to `2023.01.03` and run the flow again

xobs, 2023-01-04:
Thanks -- it does pass in `2023.01.03`. I'll post an issue on `caravel_user_project` recommending they update due to this segfault.

closed 2023-01-04 by xobs