The-OpenROAD-Project/OpenLane #1607: Add metrics generation to track QoR

vijayank88, 2023-01-06:
### Description

OpenROAD supports metrics generation to track QoR.

Command: `./openroad -metric file.json`

Implement metrics generation for OpenLane flow as well.

### Proposal

Sample `metrics.json` for floorplan stage:

```
{
	"run__flow__platform__time_units": "1ns",
	"run__flow__platform__capacitance_units": "1pF",
	"run__flow__platform__resistance_units": "1kohm",
	"run__flow__platform__voltage_units": "1v",
	"run__flow__platform__current_units": "1mA",
	"run__flow__platform__power_units": "1nW",
	"run__flow__platform__distance_units": "1um",
	"floorplan__timing__setup__tns": 0.0000,
	"floorplan__timing__setup__ws": 1.5600,
	"floorplan__power__internal__total": 0.0005,
	"floorplan__power__switching__total": 0.0001,
	"floorplan__power__leakage__total": 0.0000,
	"floorplan__power__total": 0.0006,
	"floorplan__design__io": 54,
	"floorplan__design__instance__count": 270,
	"floorplan__design__instance__area": 2284.6912,
	"floorplan__design__instance__count__stdcell": 270,
	"floorplan__design__instance__area__stdcell": 2284.6912,
	"floorplan__design__instance__count__macros": 0,
	"floorplan__design__instance__area__macros": 0.0000,
	"floorplan__design__instance__utilization": 0.0341,
	"floorplan__design__instance__utilization__stdcell": 0.0341,
	"floorplan__design__io__pre_restruct": 54,
	"floorplan__design__instance__count__pre_restruct": 270,
	"floorplan__design__instance__area__pre_restruct": 2284.6912,
	"floorplan__design__instance__count__stdcell__pre_restruct": 270,
	"floorplan__design__instance__area__stdcell__pre_restruct": 2284.6912,
	"floorplan__design__instance__count__macros__pre_restruct": 0,
	"floorplan__design__instance__area__macros__pre_restruct": 0.0000,
	"floorplan__design__instance__utilization__pre_restruct": 0.0341,
	"floorplan__design__instance__utilization__stdcell__pre_restruct": 0.0341,
	"floorplan__design__io__post_restruct": 54,
	"floorplan__design__instance__count__post_restruct": 266,
	"floorplan__design__instance__area__post_restruct": 1993.1616,
	"floorplan__design__instance__count__stdcell__post_restruct": 266,
	"floorplan__design__instance__area__stdcell__post_restruct": 1993.1616,
	"floorplan__design__instance__count__macros__post_restruct": 0,
	"floorplan__design__instance__area__macros__post_restruct": 0.0000,
	"floorplan__design__instance__utilization__post_restruct": 0.0297,
	"floorplan__design__instance__utilization__stdcell__post_restruct": 0.0297
}
```

donn, 2023-01-08:
Working on it.

dralabeing, 2023-01-25:
@donn any updates ? It will be great to sync up on METRICS support as in OR.

donn, 2023-01-25:
Nothing I can share publicly, no.

donn, 2023-04-13:
Now that OpenLane 2 is public- we do that there. We add `-metrics metrics.json` to every OpenROAD invocation and extract the results.

This issue, like many others, will remain open until OpenLane 2 is in this repository.

closed 2024-09-12 by donn