The-OpenROAD-Project/OpenLane #1617: IR Drop Report, Unconnected PDN layer: 5 & 6

Baungarten-CINVESTAV, 2023-01-12:
### Description

Hello, I am trying to integrate a small eFPGA (4 x 4 CLB) with Caravel, I have used OpenFPGA to create the FPGA architecture. 

The FPGA has been hardened and instantiated with Caravel, i just modify some variables of the caravel's config file (FP_PDN_MACRO_HOOKS, MACRO_PLACEMENT_CFG, VERILOG_FILES_BLACKBOX, EXTRA_LEFS, and EXTRA_GDS_FILES)

### Expected Behavior

[ERROR]: during executing openroad script /openlane/scripts/openroad/irdrop.tcl
[ERROR]: Log: ../home/baungarten/Proyectos_caravel/FPGA_ALL_CARAVEL/caravel_user_project/openlane/user_project_wrapper/runs/23_01_12_07_04/logs/signoff/22-irdrop.log
[ERROR]: Last 10 lines:
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2356.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2281.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2281.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2206.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2206.880um), layer: 6.
[WARNING PSM-0038] message limit reached, this message will no longer print
[WARNING PSM-0039] Unconnected instance TIE_ONE_one_ at location (68.270um, 1006.880um) layer: 5.
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
Error: irdrop.tcl, 37 PSM-0078
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...

### Environment report

```shell
How i'm working with caravel i can't run the Environment report, but i'm working with the last version of caravel. 

https://github.com/efabless/caravel_user_project
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10401865/issue_reproducible.zip)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/irdrop.tcl
[ERROR]: Log: ../home/baungarten/Proyectos_caravel/FPGA_ALL_CARAVEL/caravel_user_project/openlane/user_project_wrapper/runs/23_01_12_07_04/logs/signoff/22-irdrop.log
[ERROR]: Last 10 lines:
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2356.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2281.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2281.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2206.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1847.450um, 2206.880um), layer: 6.
[WARNING PSM-0038] message limit reached, this message will no longer print
[WARNING PSM-0039] Unconnected instance TIE_ONE_one_ at location (68.270um, 1006.880um) layer: 5.
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
Error: irdrop.tcl, 37 PSM-0078
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
```

Baungarten-CINVESTAV, 2023-01-12:
attached image
[image]

Baungarten-CINVESTAV, 2023-01-12:
Zoom out
[image]

maliberty, 2023-01-12:
That looks like a real PDN issue - @gadfort do you have time to look at this?

gadfort, 2023-01-12:
@maliberty I can try to take a look this evening

gadfort, 2023-01-14:
@Baungarten-CINVESTAV There are no followpins straps in this which is why that instance is disconnected. But you also don't have your rows cut so the follow pins will short through your block. Somehow you skipped the tapcell step which should have handled this. At present there are no issues in PDNGEN that I can tell, but there are some setup issues.

Baungarten-CINVESTAV, 2023-01-14:
Are the configuration problems in the Caravel configuration file or in the configuration file of the instantiated module?

gadfort, 2023-01-14:
@Baungarten-CINVESTAV I don't know. Maybe @maliberty can direct you to the person that can.

maliberty, 2023-01-14:
@donn this looks to be a OL issue

kareefardi, 2023-01-15:
@Baungarten-CINVESTAV By default, caravel_user_project assumes that the user area contain macros only without stdcells. `TIE_ONE_one_` and `TIE_ZERO_zero_` were inserted by `insert_tiecells`. I would wrap the constant logic in a small macro and insert it alongside `fpga_top` or have the logic inside `fpga_top` itself. If you want to have std cells along side macros, then you have to enable tap and decap insertion, fill insertion along with other things.

Baungarten-CINVESTAV, 2023-01-20:
The above error occurs when I tried to harden the user_project_w with an FPGA, the problem was that the FPGA was hardened with configuration "DESIGN_IS_CORE":1, so it was probably generating stdcells.

Now with the FPGA with configuration "DESIGN_IS_CORE":0, The FPGA harden process generates the following error.
[image]

**Reproduction material**
[Uploading issue_reproducible.zip…]()

vijayank88, 2023-01-21:
@Baungarten-CINVESTAV 
Can you share your caravel repo?

Baungarten-CINVESTAV, 2023-01-21:
Here is the repository of this project

https://github.com/Baungarten-CINVESTAV/OL_issue_1617

vijayank88, 2023-01-23:
@Baungarten-CINVESTAV 
I've cloned above repo and ran `make user_project_wrapper` and unable to find `lef/fpga_top.lef`

Baungarten-CINVESTAV, 2023-02-07:
Sorry to take much time to answer, I made multiples changes in each macro, and took much care in voltage and ground connection, finally I implemented the FPGA into the user_project_wrapper.

closed 2023-02-07 by Baungarten-CINVESTAV