The-OpenROAD-Project/OpenLane #1618: Documentation: `FP_CORE_UTIL` set too low?

xobs, 2023-01-13:
### Description

The documentation at https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/usage/hardening_macros.md#global-placement says that users should set `PL_TARGET_DENSITY` to `0.5` and `FP_CORE_UTIL` to `5`.

This would set the target density to 50% and the core utilization to 5%.

Is this correct?

### Expected Behavior

N/A

### Environment report

```shell
N/A
```

### Reproduction material

Read the documentation at https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/usage/hardening_macros.md#global-placement

### Relevant log output

```shell
N/A
```

kareefardi, 2023-04-10:
I believe that might have been intentional for "tiny" designs. @armleo Do you recall if that was correct?

kareefardi, 2024-07-17:
Documentation for `FP_CORE_UTIL` looks ok for small macros

closed 2024-07-17 by kareefardi