The-OpenROAD-Project/OpenLane #1627: LEC fails with yosys error

antonblanchard, 2023-01-16:
### Description

Commit 86615f8def77 ("Yosys Script Enhancements (#1446)") broke LEC. When building DFFRAM I see:

```
ERROR: Module `\sky130_fd_sc_hd__dlclkp_1' referenced in module `\equiv' in cell `\SLICE[3].RAM8.WORD[7].W.BYTE[0].B.genblk1.CG_gate' is not part of the design.
```

### Expected Behavior

DFFRAM builds with CONFIG_LEC enabled

### Environment report

```shell
Distribution: fedora 37
Python: v3.11.1 (OK)
Container Engine: podman v4.3.1 (UNSUPPORTED)
OpenLane Git Version: 03bbd57475d8e2de72b015a8dc5870b651844f6d
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

03bbd57 2023-01-16T23:29:39+02:00 Add a check for CURRENT_ODB to or_gui (#1625) - Matt Liberty -  (HEAD -> master, origin/master, origin/HEAD, fix-LEC)
549d7c6 2023-01-16T23:20:13+02:00 [BOT] Update PDK (#1609) - Openlane Bot -  ()
d4b42bd 2023-01-16T03:28:25+02:00 Fix loading design configs (#1623) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Test case: [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10429646/issue_reproducible.tar.gz)

### Relevant log output

```shell
ERROR: Module `\sky130_fd_sc_hd__dlclkp_1' referenced in module `\equiv' in cell `\SLICE[3].RAM8.WORD[7].W.BYTE[0].B.genblk1.CG_gate' is not part of the design.
```

donn, 2023-01-23:
Should be fixed now, I think?

kareefardi, 2023-01-25:
The fix adds `-lib` and `-setattr blackbox`. Now I know these options used to be in the script but they didn't make sense to me. I found an issue in yosys which describes the problem with that https://github.com/YosysHQ/yosys/issues/3482#issuecomment-1246571174. I believe the issue with the DFFRAM is caused due to `sky130_fd_sc_hd__dlclkp_1` being ignored by `-ignore_miss_func` and then having no definition. Blackboxing will allow reading such a cell but will only work if the two netlists have the same structure.

I think we should revert the fix and work on fixing the issues with that cell through maybe adding `function` to this cell in the liberty or making use of `equiv_*` apis

kareefardi, 2024-07-17:
LEC has been removed from OpenLane

closed 2024-07-17 by kareefardi