The-OpenROAD-Project/OpenLane #1634: LVS issues

mohamedleithy, 2023-01-22:
### Description

I had LVS issues similar to this [issue](https://github.com/The-OpenROAD-Project/OpenLane/issues/1588). I've followed the recommended adjustments and have set: ` 
"PL_RESIZER_DESIGN_OPTIMIZATIONS": 0,
"PL_RESIZER_TIMING_OPTIMIZATIONS": 0,
"CLOCK_TREE_SYNTH": 0,
"DIODE_INSERTION_STRATEGY": 0,
"FP_PDN_ENABLE_RAILS": 0,
  "FP_PDN_CHECK_NODES": 0,`. 

I'm currently getting no power ports found at the top-level (more in logs).

### Expected Behavior

Flow to complete successfully.  

### Environment report

```shell
Kernel: Darwin v21.2.0
Distribution: macOS 10.16
Python: v3.9.7 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 460d3d80dfccf66dec379a3652d79e0499298f1c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

460d3d80 2022-09-02T11:20:08+02:00 [BOT] Update yosys (#844) - Openlane Bot -  (HEAD -> master, tag: 2022.09.03)
f0f2411d 2022-09-02T11:19:30+02:00 [BOT] Update netgen (#1053) - Openlane Bot -  ()
4654b461 2022-09-01T08:52:21+02:00 Fix Tool Updater CI (Again) - Mohamed Gaber -  (tag: 2022.09.02)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[logs.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10475430/logs.zip)

### Relevant log output

```shell
[ERROR]: There are LVS errors in the design: See 'build/32x32_DEFAULT/openlane/runs/RUN_2023.01.22_20.13.07/reports/signoff/16-RAM32.lvs.rpt' for a summary and 'build/32x32_DEFAULT/openlane/runs/RUN_2023.01.22_20.13.07/logs/signoff/16-lvs.lef.log' for details.
[INFO]: Saving current set of views in 'build/32x32_DEFAULT/openlane/runs/RUN_2023.01.22_20.13.07/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'build/32x32_DEFAULT/openlane/runs/RUN_2023.01.22_20.13.07/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'build/32x32_DEFAULT/openlane/runs/RUN_2023.01.22_20.13.07/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

kareefardi, 2023-01-22:
The logs only isn't enough to fully debug the problem. Can you attach the design files and the interactive script used. Also the configuration mentioned in the attached issue may not be applicable in your case.

d-m-bailey, 2023-01-23:
@mohamedleithy #1588 is guidance on what to do for a macro only design. Your design has standard cells so you probably need  
```
"PL_RESIZER_DESIGN_OPTIMIZATIONS": 1, 
"PL_RESIZER_TIMING_OPTIMIZATIONS": 1, 
"CLOCK_TREE_SYNTH": 1, 
"DIODE_INSERTION_STRATEGY": 1, 
"FP_PDN_ENABLE_RAILS": 1, 
"FP_PDN_CHECK_NODES": 1,
```

mohamedleithy, 2023-01-23:
[reproduction.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10477093/reproduction.zip)
@kareefardi model.v is the pdk generic netlist and block_definitions.v includes the generic modules instantiated in the generic netlist. Also dffram uses a custom placer.

mohamedleithy, 2023-01-23:
@d-m-bailey still getting lvs errors with the adjustments you recommended

vijayank88, 2023-01-23:
@mohamedleithy 
Can you share your design repo?

mohamedleithy, 2023-01-23:
@vijayank88 
repo: https://github.com/AUCOHL/DFFRAM 
remote branch:  origin/add-support-to-gf180mcu 
to run:  ./dffram.py -s 32x32 -b gf180mcuC:gf180mcu_fd_sc_mcu7t5v0:ram

vijayank88, 2023-01-23:
@mohamedleithy Sorry I didn't worked with above setup.

mohamedleithy, 2023-01-23:
@vijayank88 I'm sorry I don't get what do you mean in your last comment

donn, 2023-01-23:
@mohamedleithy Vijayan has more experience with the caravel user project than DFFRAM.

mohamedleithy, 2023-01-24:
Issue was solved by adding tap cells.

closed 2023-01-24 by mohamedleithy