The-OpenROAD-Project/OpenLane #1642: Regarding formal verification in openLane rtl-to-gds flow

Brajeshp, 2023-01-29:
Request all concerned to clarify, if Formal Verification is part of openlane's RTL-To-GDS automated flow or not.

For details, one can look at starting thread, which is available at following link:-

https://app.slack.com/client/T01699QAZBQ/C016H8WJMBR/thread/C016H8WJMBR-1673179618.744079

Vinayakamk, 2023-10-09:
i wish to do a formal verification (logic equalence checks etc) for my designs.
i recently came to know about the yosys's sby and eqy tools,
but is there any move towards this in openlane's yosys regarding mentioned above?

i have same doubt regarding whether mentioned above is there in flow or not?

thank you.

kareefardi, 2024-07-17:
We don't have this as part of the flow at the moment.

closed 2024-07-17 by kareefardi