The-OpenROAD-Project/OpenLane #1643: An unexpected error 127, while running sudo make openlane

JeffreyHJF, 2023-01-30:
### Description

touch venv/created
make[1]: Entering directory '/home/eeuser/openlane/docker'
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
make[1]: /venv/bin/python3: No such file or directory
/venv/bin/python3 ./utils.py pull-if-doesnt-exist --repository efabless/openlane-tools --os centos-7 --architecture  run-base
make[1]: /venv/bin/python3: No such file or directory
make[1]: *** [Makefile:60: pull-run-base] Error 127
make[1]: Leaving directory '/home/eeuser/openlane/docker'
make: *** [Makefile:91: openlane] Error 2

### Expected Behavior

I need to install openlane successfully without error report in my terminal

### Environment report

```shell
Kernel: Linux v5.15.0-58-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/eeuser/openlane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/eeuser/openlane/dependencies/verify_versions.py", line 72, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/eeuser/openlane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (HEAD -> master, origin/master, origin/HEAD)
abb9d59 2022-07-01T13:46:02+02:00 Reimplement support for multiple power domains (#1175) - Marwan Abbas -  ()
83b6145 2022-06-26T18:19:11+02:00 Fix #1163 (#1164) - Mohamed Gaber -  ()
```

### Reproduction material

...

### Relevant log output

```shell
...
```

vijayank88, 2023-01-30:
@JeffreyHJF 
Seems trying with older release. Please follow instruction from here and try with latest OpenLane release https://github.com/The-OpenROAD-Project/OpenLane#installation-the-short-version

kareefardi, 2023-01-30:
Why are you using `sudo` ?

donn, 2023-02-02:
1. Don't use `sudo`.
2. You are using a release from July of 2022.

Prabha1010, 2023-09-30:
Step 25 (routing) failed with error:
I got this error while running make proj
Can someone help?

vijayank88, 2023-09-30:
@Prabha1010 
Please create new issue with complete log file and reproducible test case generated the run directory.