The-OpenROAD-Project/OpenLane #1645: Missing function on output ports of sky130_fd_sc_hd__(s)dlclkp_*

kareefardi, 2023-01-31:
### Description

Output ports of such cells have no function and gets rejected by yosys unless the cells are blackboxed.

### Expected Behavior

Cells shouldn't be rejected by yosys

### Environment report

```shell
n/a
```

### Reproduction material

n/a

### Relevant log output

```shell
sample output:
yosys> read_liberty libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
1. Executing Liberty frontend: libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
ERROR: Missing function on output GCLK of cell sky130_fd_sc_hd__dlclkp_1.
```

tristonw109, 2023-03-29:
I am currently having the same issue! I also have a similar issue with the nanogate45 tech

maliberty, 2023-03-29:
Fwiw I doubt yosys can infer a clock gate.

tristonw109, 2023-03-29:
> Fwiw I doubt yosys can infer a clock gate.

I need to use the insbuf command in yosy, think I can delete the clock gate section and be ok?

maliberty, 2023-03-30:
> I need to use the insbuf command in yosy, think I can delete the clock gate section and be ok?

I don't know but it seems odd to use a clock gating cell as a buffer.

tristonw109, 2023-03-30:
> Fwiw I doubt yosys can infer a clock gate.

Using this comment, I looked into if yosys can infer gated clocks. It doesn't seem to support it. Since I didn't need the gate clock, I removed it from the liberty file and was able to easily load it into yosys. This is a temporary fix until yosys can support gated clocks.

pramchan, 2025-06-09:
Any reason for blocking this? I had similar yosys> read_liberty libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
1. Executing Liberty frontend: libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
ERROR: Missing function on output GCLK of cell sky130_fd_sc_hd__dlclkp_1. - Any updates?

donn, 2025-10-04:
@pramchan Oh, sorry for the late repsonse.

If you're stuck still, Yosys has a number of new arguments to `read_liberty` which you can use `-ignore_miss_func` to work around this specific issue.

https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/index_frontends.html#cmd-read_liberty

The reason it is blocked is that it's not technically an OpenLane bug and I'm not sure why it's here.