The-OpenROAD-Project/OpenLane #1648: [ERROR GRT-0118] Routing congestion too high.

MihaiHMO, 2023-02-03:
### Description

Hi ,
I am trying to do the PNR for a small design and I get stuck at routing.
I get this error : [ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI. or Error: resizer_routing_timing.tcl, 39 GRT-0118 -
How do I interpret the GUI view? It is really tight if I use "PL_TARGET_DENSITY": 0.07 , and FP_CORE_UTIL : 7 ?
Or what I need cover before %run_routing in the previous steps? All previous steps are ok, no error reported.
I saw also some discussions on analog macros. I have also a SRAM which I just imported (lef, verilog, gds) , not generated from scratch .
If I open the SRAM design separately in magic I do not get any DRC. I rtied to solve it on slack channel but nobody could help. 
 

### Expected Behavior

To pass the routing stage. 

### Environment report

```shell
Kernel: Linux v5.15.0-58-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.23 (OK)
OpenLane Git Version: c60657c6637349cc7b0b64b36dcd3c0fb7962a10
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c60657c 2023-01-31T16:03:00+02:00 [BOT] Update PDK (#1644) - Openlane Bot -  (grafted, HEAD -> master, tag: 2023.02.01, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10579864/issue_reproducible.zip)
[issue_reproducible.z01.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10579881/issue_reproducible.z01.zip)
I renamed the file issue_reproducible.z01 , please rename it back.

### Relevant log output

```shell
[STEP 11]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: designs/vsdmemsoc/runs/RUN_UW/logs/routing/11-resizer.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: designs/vsdmemsoc/runs/RUN_UW/logs/routing/11-resizer.log
[ERROR]: Last 10 lines:

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 64556
[INFO GRT-0198] Via related Steiner nodes: 4644
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 81777
[INFO GRT-0112] Final usage 3D: 490825
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_timing.tcl, 39 GRT-0118
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager
```

vijayank88, 2023-02-03:
@habibayassin 
Is that extraction correct?
```
Archive:  issue_reproducible.zip
warning [issue_reproducible.zip]:  zipfile claims to be last disk of a multi-part archive;
  attempting to process anyway, assuming all parts have been concatenated
  together in order.  Expect "errors" and warnings...true multi-part support
  doesn't exist yet (coming soon).
file #1:  bad zipfile offset (local header sig):  4
file #2:  bad zipfile offset (local header sig):  81
file #3:  bad zipfile offset (local header sig):  1526
file #4:  bad zipfile offset (local header sig):  1607
file #5:  bad zipfile offset (local header sig):  1696
file #6:  bad zipfile offset (local header sig):  1794
file #7:  bad zipfile offset (local header sig):  1908
file #8:  bad zipfile offset (local header sig):  2026
file #9:  bad zipfile offset (local header sig):  2223208
file #10:  bad zipfile offset (local header sig):  4540555
file #11:  bad zipfile offset (lseek):  19202048
file #12:  bad zipfile offset (EOF):  19208560
file #13:  bad zipfile offset (EOF):  19208641
file #14:  bad zipfile offset (EOF):  19208732
  inflating: issue_reproducible/tmp/merged.nom.lef  
   creating: issue_reproducible/openlane/
   creating: issue_reproducible/openlane/designs/
   creating: issue_reproducible/openlane/designs/vsdmemsoc/
   creating: issue_reproducible/openlane/designs/vsdmemsoc/src2/
   creating: issue_reproducible/openlane/designs/vsdmemsoc/src2/lib/
  inflating: issue_reproducible/openlane/designs/vsdmemsoc/src2/lib/sram_32_256_sky130A_TT_1p8V_25C.lib  
   creating: issue_reproducible/openlane/scripts/
   creating: issue_reproducible/openlane/scripts/openroad/
   creating: issue_reproducible/openlane/scripts/openroad/common/
  inflating: issue_reproducible/openlane/scripts/openroad/common/set_global_connections.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/common/set_routing_layers.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/common/io.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/common/set_layer_adjustments.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/common/resizer.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/common/dpl_cell_pad.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/common/set_rc.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/sta.tcl  
  inflating: issue_reproducible/openlane/scripts/openroad/resizer_routing_timing.tcl  
  inflating: issue_reproducible/run.tcl  
   creating: issue_reproducible/results/
   creating: issue_reproducible/results/placement/
  inflating: issue_reproducible/results/placement/vsdmemsoc.nl.v  
   creating: issue_reproducible/results/cts/
  inflating: issue_reproducible/results/cts/vsdmemsoc.def  
  inflating: issue_reproducible/results/cts/vsdmemsoc.sdc  
  inflating: issue_reproducible/results/cts/vsdmemsoc.odb  
  inflating: issue_reproducible/env.gdb  
```

maliberty, 2023-02-03:
I don't understand you comment about renaming the issue file.  I tried issue_reproducible.zip but I get:

Error: resizer_routing_timing.tcl, 15 cannot read file ./tmp/synthesis/resizer_sky130_fd_sc_hd__tt_025C_1v80.lib.

MihaiHMO, 2023-02-03:
After splitin the zip file I got 2 files:
issue_reproducible.zip
issue_reproducible.z01
Github did not allow me to upload  issue_reproducible.z01 so I renamed  it issue_reproducible.z01.zip so you need to change it back.

maliberty, 2023-02-03:
how did you split it?

vijayank88, 2023-02-04:
I think they followed:
```
    Warning:
    GitHub's maximum file upload size is 25 MB, so if your upload is larger than 25MB, please consider splitting it as follows:
    for tar files: split -b 24M <tar-file-name> "<file-tar-name>.part"
    for zip files: zip -r -s 24M <zip-file-name> <directory-name>
```

vijayank88, 2023-02-04:
Routing congestion related to sram macro. Got reply from @MihaiHMO `By sram cell with  sky130_sram_1kbyte_1rw1r_32x256_8 from PDK , I pass routing.`

maliberty, 2023-02-04:
can this be closed then?

MihaiHMO, 2023-02-04:
Yes

closed 2023-02-04 by maliberty