The-OpenROAD-Project/OpenLane #1649: [ERROR GRT-0118] Routing congestion too high.

MihaiHMO, 2023-02-03:
### Description

Hi ,
I am trying to do the PNR for a small design and I get stuck at routing.
I get this error : [ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI. or Error: resizer_routing_timing.tcl, 39 GRT-0118 -
How do I interpret the GUI view? It is really tight if I use "PL_TARGET_DENSITY": 0.07 , and FP_CORE_UTIL : 7 ?
Or what I need cover before %run_routing in the previous steps? All previous steps are ok, no error reported.
I saw also some discussions on analog macros. I have also a SRAM which I just imported (lef, verilog, gds) , not generated from scratch .
If I open the SRAM design separately in magic I do not get any DRC. I rtied to solve it on slack channel but nobody could help. 
 

### Expected Behavior

To pass the routing stage. 

### Environment report

```shell
Kernel: Linux v5.15.0-58-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.23 (OK)
OpenLane Git Version: c60657c6637349cc7b0b64b36dcd3c0fb7962a10
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c60657c 2023-01-31T16:03:00+02:00 [BOT] Update PDK (#1644) - Openlane Bot -  (grafted, HEAD -> master, tag: 2023.02.01, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.z01.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10579924/issue_reproducible.z01.zip)
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10579927/issue_reproducible.zip)

### Relevant log output

```shell
[STEP 11]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: designs/vsdmemsoc/runs/RUN_UW/logs/routing/11-resizer.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: designs/vsdmemsoc/runs/RUN_UW/logs/routing/11-resizer.log
[ERROR]: Last 10 lines:

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 64556
[INFO GRT-0198] Via related Steiner nodes: 4644
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 81777
[INFO GRT-0112] Final usage 3D: 490825
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_timing.tcl, 39 GRT-0118
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager
```

maliberty, 2023-02-03:
Is this a duplicate of #1648 ?

MihaiHMO, 2023-02-03:
Yes, sorry, for #1649 I got an error after pressing submit.

donn, 2023-02-03:
Closing as a duplicate- will follow up in the original.

closed 2023-02-03 by donn