The-OpenROAD-Project/OpenLane #1657: Error for last steps in openlane flow

MihaiHMO, 2023-02-08:
### Description

Fail at the final steps. 

### Expected Behavior

Not to fail

### Environment report

```shell
Kernel: Linux v5.15.0-58-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v23.0.0 (OK)
OpenLane Git Version: c60657c6637349cc7b0b64b36dcd3c0fb7962a10
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c60657c 2023-01-31T16:03:00+02:00 [BOT] Update PDK (#1644) - Openlane Bot -  (grafted, HEAD -> master, tag: 2023.02.01, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

I do not have a reproducible package. 
But I can attache some log files.
[openlane_logs.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10683678/openlane_logs.zip)

### Relevant log output

```shell
[ERROR]: during executing: "python3 /openlane/scripts/generate_reports.py -d /openlane/designs/vsdmemsoc --design_name vsdmemsoc --tag RUN_2023.02.07_15.41.57 --output_file /openlane/designs/vsdmemsoc/runs/RUN_2023.02.07_15.41.57/reports/metrics.csv --man_report /openlane/designs/vsdmemsoc/runs/RUN_2023.02.07_15.41.57/reports/manufacturability.rpt --run_path /openlane/designs/vsdmemsoc/runs/RUN_2023.02.07_15.41.57"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/generate_reports.py", line 50, in cli
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openlane/scripts/report/report.py", line 775, in get_report
    *self.extract_all_values(),
  File "/openlane/scripts/report/report.py", line 514, in extract_all_values
    for line in klayout_violations.split("\n"):
AttributeError: 'int' object has no attribute 'split'

[ERROR]: during executing: "python3 /openlane/scripts/generate_reports.py -d /openlane/designs/vsdmemsoc --design_name vsdmemsoc --tag RUN_2023.02.07_15.41.57 --output_file /openlane/designs/vsdmemsoc/runs/RUN_2023.02.07_15.41.57/reports/metrics.csv --man_report /openlane/designs/vsdmemsoc/runs/RUN_2023.02.07_15.41.57/reports/manufacturability.rpt --run_path /openlane/designs/vsdmemsoc/runs/RUN_2023.02.07_15.41.57"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/generate_reports.py", line 50, in cli
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openlane/scripts/report/report.py", line 775, in get_report
    *self.extract_all_values(),
  File "/openlane/scripts/report/report.py", line 514, in extract_all_values
    for line in klayout_violations.split("\n"):
AttributeError: 'int' object has no attribute 'split'

[ERROR]: Flow failed.
```

vijayank88, 2023-02-08:
@MihaiHMO 
Latest commit is https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2023.02.08.
Can you pull the latest commit and try again and confirm is same error repeating?

MihaiHMO, 2023-02-08:
how I do this? I need to reinstall ? 
additional... one day ago it work - here you have a log from that run
[openlane.log](https://github.com/The-OpenROAD-Project/OpenLane/files/10684231/openlane.log)
ed.

vijayank88, 2023-02-08:
Run below steps to install latest commits: https://openlane.readthedocs.io/en/latest/getting_started/updating.html#updating-openlane
```
cd OpenLane
git checkout master
git pull
make
```
Run your design again and check the status

vijayank88, 2023-02-08:
From your `openlane.log`
```
[ERROR]: There are hold violations in the design at the typical corner. Please refer to 'designs/vsdmemsoc/runs/RUN_FULL_10/reports/signoff/27-rcx_sta.min.rpt'.
```
Please fix the hold violation issue to complete the flow.

MihaiHMO, 2023-02-08:
> x the hold violation issue to complete the flow.
[openlane_logs230208_2.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10684714/openlane_logs230208_2.zip)

done... same errors

vijayank88, 2023-02-08:
In that case we need your design folder(vsdmemsoc) to reproduce the issue. Attach only source files/config.json

vijayank88, 2023-02-08:
@kareefardi fyi

kareefardi, 2023-02-08:
Looking at the logs, there is a small bug in `report.py` which I think is simple enough to fix. I believe this was triggered by enabling `RUN_KLAYOUT_DRC`. Since klayout drc is not working at the moment, I suggest disabling it anyway

MihaiHMO, 2023-02-08:
yes this is the issue. I got the info that magic does not work with SRAM macro and was recomended to do DRC in klayout. so I tested this . we can close the ticket. thank you.

closed 2023-02-13 by donn