The-OpenROAD-Project/OpenLane #1659: Routing Fails

mohamedleithy, 2023-02-14:
### Description

Routing fails 

### Expected Behavior

Flow to complete

### Environment report

```shell
Kernel: Linux v5.15.0-60-generic
Distribution: ubuntu 20.04
Python: v3.8.15 (OK)
Container Engine: docker v20.10.19 (OK)
OpenLane Git Version: c40bcaca582bb192606265d49a582fb9ee3ba1be
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c40bcac 2022-09-22T07:26:28+02:00 [BOT] Update PDK (#1389) - Openlane Bot -  (HEAD -> master, tag: 2022.09.23, origin/master, origin/HEAD)
c8f917e 2022-09-22T07:23:38+02:00 Correctly decorate wrapper in click_odb (#1381) - Matt Liberty -  ()
f81bd81 2022-09-22T07:21:39+02:00 Correct the installation link (#1384) - Arman Avetisyan -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10730569/issue_reproducible.zip)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Log: build/256x32_DEFAULT/openlane/runs/RUN_2023.02.12_15.06.10/logs/routing/4-detailed.log
[ERROR]: Last 10 lines:
    elapsed time = 00:08:44, memory = 48704.07 (MB).
    Completing 20% with 134900 violations.
    elapsed time = 00:09:09, memory = 48730.28 (MB).
    Completing 30% with 131053 violations.
    elapsed time = 00:09:37, memory = 48677.89 (MB).
    Completing 40% with 131053 violations.
    elapsed time = 00:18:27, memory = 48813.32 (MB).
    Completing 50% with 131053 violations.
    elapsed time = 00:18:27, memory = 48813.73 (MB).
child killed: kill signal

[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.
```

vijayank88, 2023-02-14:
@mohamedleithy 
Child killed is a memory issue.
Can you increase your swap space and try?
It is used close to 50GB RAM. What is available RAM and core?

vijayank88, 2023-02-14:
Also this is month old tag: 2022.09.23. Update to latest one and try. Coz there are updates related PDK as well.

vijayank88, 2023-02-15:
@mohamedleithy 
Placement density looks high. I ran locally and end up with 60k DRC violations.
Please re-floorplan and relax placement density and try again.

closed 2023-02-15 by mohamedleithy