The-OpenROAD-Project/OpenLane #1662: flow.tcl fails with TERM=dumb

proppy, 2023-02-18:
### Description

`flow.tcl` produce the following error when `TERM` is set to `dumb`
```
child process exited abnormally
    while executing
"exec tput setaf $color"
```

### Expected Behavior

`flow.tcl` fallback to uncolored output, or expose a flag for turning off rich output.

### Environment report

```shell
~/silicon-env/share/openlane/env.py issue-survey
Traceback (most recent call last):
  File "/home/proppy/silicon-env/share/openlane/env.py", line 265, in <module>
    main()
  File "/home/proppy/silicon-env/share/openlane/env.py", line 261, in main
    commands[args[0]]()
  File "/home/proppy/silicon-env/share/openlane/env.py", line 87, in issue_survey
    os_info = OSInfo.get()
  File "/home/proppy/silicon-env/share/openlane/dependencies/env_info.py", line 160, in get
    key, value = line.split("=")
ValueError: not enough values to unpack (expected 2, got 1)
```
```

### Reproduction material

N/A

### Relevant log output

```shell
(base) openlane 🍊 flow.tcl -design inverter
OpenLane 2023.01.09_1_gac83c37-conda
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

child process exited abnormally
    while executing
"exec tput setaf $color"
    (procedure "color_text" line 3)
    invoked from within
"color_text 3 "$message""
    (procedure "puts_warn" line 3)
    invoked from within
"puts_warn "OpenLane may not function properly: $::env(VCHECK_OUTPUT)""
    invoked from within
"if [catch {exec python3 $::env(OPENLANE_ROOT)/dependencies/verify_versions.py $arg_values(-test_mismatches)} ::env(VCHECK_OUTPUT)] {
        if { ![in..."
    (procedure "prep" line 33)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    (file "/home/proppy/silicon-env/share/openlane/flow.tcl" line 422)
    invoked from within
"source "$::env(CONDA_PREFIX)/share/openlane/flow.tcl""
    (file "/home/proppy/silicon-env/bin/flow.tcl" line 3)
```

closed 2023-02-18 by donn

proppy, 2023-02-21:
Does it make sense to also handle `TERM=unknown` the same way?

donn, 2023-02-21:
I'd say so yeah

PR #1663 (2023-02-18): utils: handle `TERM=dumb`
Fixes #1662