The-OpenROAD-Project/OpenLane #1674: save_views in scripts/tcl_commands/all.tcl leaves compressed files

d-m-bailey, 2023-02-28:
### Description

In the `save_views` function of `scripts/tcl_commands/all.tcl`, files are copied to destination directories.

However, compressed versions of the file may already exist in those directories.

efabless' `caravel` `Makefile`, for example, runs `make uncompress` before integrating data. There is the possibility that an older compressed version of the data will overwrite the newer uncompressed version.

### Proposal

Have `save_views` remove any compressed versions of the data in the destination directory.

donn, 2023-02-28:
Too... magical perhaps? Shouldn't `make compress` in Caravel be updated?

RTimothyEdwards, 2023-02-28:
We should stop uncompressing and compressing files, at least the ones that all the tools can handle in compressed form (like GDS).  "make compress"/"make uncompress" should only be splitting or merging these files to avoid the repository size limit.

d-m-bailey, 2023-02-28:
@RTimothyEdwards Can magic handle links to compressed files in mag files now?

@donn unconditionally deleting files bothers me too. On the other hand, I also feel that the output directory should contain the data unambiguously.

donn, 2023-03-21:
This is entirely on Caravel, sorry. I don't see anything I can do here that wouldn't be magical, pun not intended.

closed 2023-03-21 by donn