The-OpenROAD-Project/OpenLane #1677: Flow failed at resize stage of LDPC project

JiaDYuan, 2023-03-09:
### Description

I was using OpenLane to try to pass the VLSI flow of LDPC project. This LDPC is pure logic without memory needing to be dropped in sram. But I failed in the stage of resize. Here are the error messages:

[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: designs/ldpc_4x4_serial/runs/LDPC/logs/placement/8-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  ldpc_34._87060_
[INFO DPL-0035]  ldpc_34._79598_
[INFO DPL-0035]  ldpc_34._87052_
[INFO DPL-0035]  ldpc_34._88702_
[INFO DPL-0035]  ldpc_34._80460_
[INFO DPL-0035]  ldpc_34._80460_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 62 DPL-0036
child process exited abnormally

[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.

### Expected Behavior

I want to pass the VLSI flow of the LDPC project in OpenLane.

### Environment report

```shell
Python 3.10.9
jiadyuan@JiadongdeMBP OpenLane % python3 ./env.py issue-survey
Kernel: Darwin v22.2.0
Distribution: macOS 13.1
Python: v3.10.9 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: a664c0e162fe8c8308f1e63ab3ebdea109c2c4ee
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a664c0e 2023-03-02T13:15:42+02:00 More strict antenna violations repair (#1671) - Mohamed Hosni -  (HEAD -> master, tag: 2023.03.03, origin/master, origin/HEAD)
93c8d84 2023-03-02T12:55:32+02:00 Add repair_design for global routing (#1670) - Mohamed Hosni -  ()
e10820e 2023-02-28T18:26:21+02:00 Add PL_WIRELENGTH_COEF to global placement. (#1672) - Mohamed Hosni -  (tag: 2023.03.01)
---
Git Remotes

origin	http://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	http://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[reproducible.z02.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933752/reproducible.z02.zip)
[reproducible.z03.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933756/reproducible.z03.zip)
[reproducible.z04.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933761/reproducible.z04.zip)
[reproducible.z06.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933770/reproducible.z06.zip)
[reproducible.z07.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933772/reproducible.z07.zip)
[reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933783/reproducible.zip)
[reproducible.z01.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933791/reproducible.z01.zip)[reproducible.z05.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10933802/reproducible.z05.zip)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: designs/ldpc_4x4_serial/runs/LDPC/logs/placement/8-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  ldpc_34._87060_
[INFO DPL-0035]  ldpc_34._79598_
[INFO DPL-0035]  ldpc_34._87052_
[INFO DPL-0035]  ldpc_34._88702_
[INFO DPL-0035]  ldpc_34._80460_
[INFO DPL-0035]  ldpc_34._80460_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 62 DPL-0036
child process exited abnormally

[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.
```

vijayank88, 2023-03-10:
@JiaDYuan 
add following variable in your `config.json` and try again.
```
DPL_CELL_PADDING : 2,
```

maliberty, 2023-03-11:
I tried to download and reassemble your zip files but was unsuccessful.  Would you provide the steps necessary?

vijayank88, 2023-03-11:
I also struggle a lot to merge and create single zip file and unzip again.

Found only 7zip is good to unzip multi-part compressed file. Used following command:
```
cat reproducible.z0* > reproducible.zip
7za x reproducible.zip
```

maliberty, 2023-03-11:
Your design is unconstrained.  In set ::env(CURRENT_SDC) {./tmp/floorplan/3-initial_fp.sdc};

```
create_clock -name clk -period 10.0000 
set_clock_uncertainty 0.2500 clk
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {clock}]
```
The clock signal is not being treated as an input not as a clock.   You can see this in
```
>>> check_setup
Warning: There are 32768 unclocked register/latch pins.
Warning: There are 34832 unconstrained endpoints.
```

Also the design seems very dense.  What are you using for global placement density and padding?

JiaDYuan, 2023-03-12:
> @JiaDYuan add following variable in your `config.json` and try again.
> 
> ```
> DPL_CELL_PADDING : 2,
> ```
This is my Config.json:
{
    "DESIGN_NAME": "ldpc_4x4_serial",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 10.0,
    "SYNTH_NO_FLAT": true,
    "SYNTH_STRATEGY": "DELAY 4",
    "DESIGN_IS_CORE": false,
    "DPL_CELL_PADDING" : 2,
    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 2000 3750",
    "PL_TARGET_DENSITY": 0.5
}

I re-run and got the same messages:
[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: designs/ldpc_4x4_serial/runs/LDPC/logs/placement/10-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  ldpc_34._79525_
[INFO DPL-0035]  ldpc_34._79525_
[INFO DPL-0035]  ldpc_34._79525_
[INFO DPL-0035]  ldpc_34._77299_
[INFO DPL-0035]  ldpc_12._77230_
[INFO DPL-0035]  ldpc_34._79974_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 62 DPL-0036
child process exited abnormally

JiaDYuan, 2023-03-12:
> Your design is unconstrained. In set ::env(CURRENT_SDC) {./tmp/floorplan/3-initial_fp.sdc};
>
> ```
> create_clock -name clk -period 10.0000
> set_clock_uncertainty 0.2500 clk
> set_input_delay 2.0000 [...]

How to set the clock?

And This is my Config.json:
{
"DESIGN_NAME": "ldpc_4x4_serial",
"VERILOG_FILES": "dir::src/*.v",
"CLOCK_PORT": "clk",
"CLOCK_PERIOD": 10.0,
"SYNTH_NO_FLAT": true,
"SYNTH_STRATEGY": "DELAY 4",
"DESIGN_IS_CORE": false,
"DPL_CELL_PADDING" : 2,
"FP_SIZING": "absolute",
"DIE_AREA": "0 0 2000 3750",
"PL_TARGET_DENSITY": 0.5
}

JiaDYuan, 2023-03-12:
And I have a question,  how to set the parameters correctly? eg. DIE_AREA, DENSITY, CLOCK, etc... Is there a detailed tutorial or manual?

maliberty, 2023-03-12:
@donn I see this in the pdk setup:
```
pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/config.tcl:set ::env(GPL_CELL_PADDING) {0}
pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/config.tcl:set ::env(DPL_CELL_PADDING) {4}
```

It is nonsensical to have DPL_CELL_PADDING > GPL_CELL_PADDING.  I think this is a serious setup issue that goes beyond this particular test case.

JiaDYuan, 2023-03-13:
Hi, now I have  modified my config as below:
{
    "DESIGN_NAME": "ldpc_4x4_serial",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 10.0,
    "SYNTH_NO_FLAT": true,
    "SYNTH_STRATEGY": "DELAY 4",
    "DESIGN_IS_CORE": false,
    "DPL_CELL_PADDING" : 2,
    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 4000 6750",
    "PL_TARGET_DENSITY": 0.5
}

And now it is stuck in the Global Routing Resizer Design Optimizations.

 report_design_area
============================================================================
Design area 9913987 u^2 37% utilization.
area_report_end
Setting global connections for newly added cells...
Writing OpenROAD database to /openlane/designs/ldpc_4x4_serial/runs/LDPC/tmp/placement/8-resizer.odb...
Writing netlist to /openlane/designs/ldpc_4x4_serial/runs/LDPC/tmp/placement/8-resizer.nl.v...
Writing powered netlist to /openlane/designs/ldpc_4x4_serial/runs/LDPC/tmp/placement/8-resizer.pnl.v...
Writing layout to /openlane/designs/ldpc_4x4_serial/runs/LDPC/tmp/placement/8-resizer.def...
Writing timing constraints to /openlane/designs/ldpc_4x4_serial/runs/LDPC/tmp/placement/8-resizer.sdc...

JiaDYuan, 2023-03-13:
Ok, finally the process is finished with error messages:
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: designs/ldpc_4x4_serial/runs/LDPC/logs/routing/12-resizer_design.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 4018502
[INFO GRT-0198] Via related Steiner nodes: 420215
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 8385862
[INFO GRT-0112] Final usage 3D: 55745682
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_design.tcl, 39 GRT-0118
child process exited abnormally

[ERROR]: Creating issue reproducible...

The congestion is too high, so how should I configure the parameters?

[ldpc.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10955503/ldpc.zip)

vijayank88, 2023-03-13:
@JiaDYuan As per ldpc verilog you attached, defined wrong clock port/net name in the json file.
Replace `clk` with `clock` as per source RTL in `config.json`

JiaDYuan, 2023-03-13:
[reproducible.z02.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960643/reproducible.z02.zip)
[reproducible.z03.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960646/reproducible.z03.zip)
[reproducible.z04.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960665/reproducible.z04.zip)
[reproducible.z05.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960668/reproducible.z05.zip)
[reproducible.z06.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960673/reproducible.z06.zip)
[reproducible.z07.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960676/reproducible.z07.zip)
[reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960679/reproducible.zip)
[reproducible.z01.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960682/reproducible.z01.zip)

FYI, Github doesn't support incompatible suffix, so before unzip the files, just remove the .zip suffix out of the files from reproducible.z01.zip to reproducible.z07.zip. Then use command: zip issue_reproducible.zip -s=0 --out reproducible.zip

Now, I got the CTS errors:

[ERROR]: during executing openroad script /openlane/scripts/openroad/cts.tcl
[ERROR]: Log: designs/ldpc_4x4_serial/runs/LDPC/logs/cts/10-cts.log
[ERROR]: Last 10 lines:
[INFO CTS-0004] Total number of Buffers Inserted: 5850.
[INFO CTS-0005] Total number of Clock Subnets: 5850.
[INFO CTS-0006] Total number of Sinks: 32768.
cts_report_end
min_report

===========================================================================
report_checks -path_delay min (Hold)
============================================================================
child killed: kill signal

antonblanchard, 2023-03-13:
> child killed: kill signal

Did you run out of memory?

JiaDYuan, 2023-03-13:
> > child killed: kill signal
> 
> Did you run out of memory?

I am not sure. How could I check? I was running on MacOS.

vijayank88, 2023-03-14:
I think the `reg 2047:0` consume more resource and getting killed.
I've used your attached and got killed during cts resizer.

JiaDYuan, 2023-03-14:
> 

So, is it possible? I know the LDPC got about 1 millions gates, how many memories do I  need to pass the VLSI flow?

vijayank88, 2023-03-14:
> [reproducible.z02.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10960643/reproducible.z02.zip) [...]

I have ran this test case with GCP cloud (128GB RAM/32 Core). Able to pass cts resizer stage without any issue.
```
Design area 9721679 u^2 58% utilization.
area_report_end
Setting global connections for newly added cells...
Writing OpenROAD database to ./out.odb...
Writing layout to ./out.def...
Writing timing constraints to ./results/cts/ldpc_4x4_serial.sdc...
openroad> 
```
Can you update OpenLane commit and try again?

donn, 2023-03-14:
> It is nonsensical to have DPL_CELL_PADDING > GPL_CELL_PADDING. I think this is a serious setup issue that goes beyond this particular test case.

@maliberty As I recall, we found `GPL_CELL_PADDING` redundant given the density controls. Does it cause issues?

donn, 2023-03-15:
Hmm. I'll discuss with the team. Thanks for the heads up.

JiaDYuan, 2023-03-15:
> I have ran this test case with GCP cloud (128GB RAM/32 Core). Able to pass cts resizer stage without any issue.
>
> ```
> Design area 9721679 u^2 58% utilization.
> area_report_end
> Setting global [...]

Hi,  I updated OpenLane and tried again, it's still the sam cts issue...

maliberty, 2023-03-15:
I suggest you try setting gpl padding to the same as dpl padding

maliberty, 2023-03-15:
Tell OL to update if that is the case.  You can also locally update the OR version in OL if needed.

vijayank88, 2023-03-16:
@JiaDYuan 
Is that possible to share your updated `config.json` and `src` files.?

JiaDYuan, 2023-03-16:
> @JiaDYuan
> Is that possible to share your updated `config.json` an
[source_and_config.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10992768/source_and_config.zip)
d `src` files.?

JiaDYuan, 2023-03-16:
I was setting GPL as same as DPL:  "DPL_CELL_PADDING" : 2,    "GPL_CELL_PADDING" : 2. And I got the same CTS errors:

[INFO]: Running Clock Tree Synthesis (log: designs/ldpc_4x4_serial/runs/LDPC/logs/cts/10-cts.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/cts.tcl
[ERROR]: Log: designs/ldpc_4x4_serial/runs/LDPC/logs/cts/10-cts.log
[ERROR]: Last 10 lines:
[INFO CTS-0004] Total number of Buffers Inserted: 5762.
[INFO CTS-0005] Total number of Clock Subnets: 5762.
[INFO CTS-0006] Total number of Sinks: 32768.
cts_report_end
min_report

===========================================================================
report_checks -path_delay min (Hold)
============================================================================
child killed: kill signal

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...

maliberty, 2023-03-16:
Would you attach the file generated by "[ERROR]: Creating issue reproducible..."

Is this an out of memory issue?

JiaDYuan, 2023-03-16:
[reproducible.z02.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996446/reproducible.z02.zip)
[reproducible.z03.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996457/reproducible.z03.zip)
[reproducible.z04.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996466/reproducible.z04.zip)
[reproducible.z05.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996471/reproducible.z05.zip)
[reproducible.z06.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996477/reproducible.z06.zip)
[reproducible.z07.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996484/reproducible.z07.zip)
[reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996493/reproducible.zip)
[reproducible.z01.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10996498/reproducible.z01.zip)

@maliberty 
Please remove suffix .zip from 01 to 07, then unzip reproducible.

vijayank88, 2023-03-17:
@donn Can you share your system resources like total RAM and no. of cores?

JiaDYuan, 2023-03-17:
@vijayank88 
CPU :  2,3 GHz 4-core Intel Core i7
GPU :  Intel Iris Plus Graphics 1536 MB
Memory : 16 GB 3733 MHz LPDDR4X
macOS : 13.2.1 (22D68)

vijayank88, 2023-03-17:
`child killed: kill signal` is most of the time out of memory issue.
Increase swap memory and try again the flow.
At my end cts stage successfully completed.
```
Design area 9710149 u^2 58% utilization.
area_report_end
Setting global connections for newly added cells...
Writing OpenROAD database to ./out.odb...
Writing layout to ./out.def...
Writing timing constraints to ./results/cts/ldpc_4x4_serial.sdc...
openroad> 
```

antonblanchard, 2023-03-17:
> Memory : 16 GB 3733 MHz LPDDR4X

I doubt that will be enough for a design with a million gates. You are almost certainly going to need to cut down your design or increase your memory.

maliberty, 2023-03-18:
I agree - I don't see any further action here from OR.  Please reopen if you get more memory and still have trouble.

closed 2023-03-18 by maliberty

vijayank88, 2023-03-18:
@JiaDYuan 
Another suggestion try to harden the `ldpc` as a macro and integrate to top level.

JiaDYuan, 2023-03-20:
> @JiaDYuan Another suggestion try to harden the `ldpc` as a macro and integrate to top level.

@vijayank88 @maliberty 
Hi, I migrated the whole project to a server with 110 GB memory and 168 GB swap mem.  But it was stuck in Placement Resizer Timing Optimizations  for more than one day. It just kept running without messages, so I terminated it and tried to  harder the ldpc as a macro first. But I got this error message:

[ERROR PPL-0024] Number of pins 4100 exceeds max possible 2519.
Error: ioplacer.tcl, 51 PPL-0024
child process exited abnormally

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11016814/issue_reproducible.zip)

So how can I set up the numbers of IO pins?

vijayank88, 2023-03-20:
@JiaDYuan 
Considering issue already closed, please open new issue always. Don't merge different issue under one title.
Next time follow that.
Use `"FP_IO_MIN_DISTANCE": 1,` in `config.json` and try. If not fixed you need to increase the die size.