The-OpenROAD-Project/OpenLane #1682: PDN creating stripes of met4 over macros too close to existing power straps

mattvenn, 2023-03-16:
### Description

precheck is failing due to DRC on met4 spacing.

[image]

which results in things like this:

[image]

### Expected Behavior

vertical met4 power stripes should not run over macros

### Environment report

```shell
Kernel: Linux v5.13.0-28-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

https://github.com/TinyTapeout/tinytapeout-03
make uncompress
make run-precheck

### Relevant log output

```shell
{{FINISH}} Executing Finished, the full log 'precheck.log' can be found in '/home/matt/work/asic-workshop/shuttle9/caravel_user_project/precheck_results/16_MAR_2023___09_36_59/logs'  
{{FAILURE}} 1 Check(s) Failed: ['Klayout BEOL'] !!!
```

mattvenn, 2023-03-16:
my workaround is to disable vertical straps in the pdn_cfg.tcl

maliberty, 2023-03-16:
@gadfort any thoughts?

gadfort, 2023-03-16:
@maliberty not off the top of my head. I would have to run it to get a better feel for the issue.

maliberty, 2023-03-18:
I can't reproduce from the instructions given

```
% make uncompress
export CARAVEL_ROOT=/home/mliberty/bugs/ol/1682/tinytapeout-03-main/caravel && make -f /home/mliberty/bugs/ol/1682/tinytapeout-03-main/caravel/Makefile uncompress
Makefile:29: warning: undefined variable 'PDK_ROOT'
make[1]: Entering directory '/home/mliberty/bugs/ol/1682/tinytapeout-03-main'
make[1]: /home/mliberty/bugs/ol/1682/tinytapeout-03-main/caravel/Makefile: No such file or directory
make[1]: *** No rule to make target '/home/mliberty/bugs/ol/1682/tinytapeout-03-main/caravel/Makefile'.  Stop.
make[1]: Leaving directory '/home/mliberty/bugs/ol/1682/tinytapeout-03-main'
make: *** [Makefile:84: uncompress] Error 2
```

A self-contained test case would be preferrable.

mattvenn, 2023-03-20:
I'm not sure how to do a self contained test for this. @kareefardi any ideas?
@maliberty you'd have to install caravel submodule with make install,
and have PDK_ROOT and OPENLANE_ROOT set to where pdk and openlane are installed.

vijayank88, 2023-03-20:
@mattvenn 
Please add `exit 1` in https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/pdn.tcl#L33 and run `make user_project_wrapper`. It will produce the `issue_reproducible` and attach it here.
Then restore `pdn.tcl` to original state.

mattvenn, 2023-03-20:
[reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11017955/reproducible.tar.gz)

maliberty, 2023-03-20:
@mattvenn is this urgent?  If not I'll leave it for Peter

mattvenn, 2023-03-21:
Not urgent, I resolved  it by turning off vertical power straps.
Matt

kareefardi, 2024-07-17:
I no longer see such nets. Please open a new issue if the problem persists.

closed 2024-07-17 by kareefardi