The-OpenROAD-Project/OpenLane #1689: docopt not in requirements.txt

mguthaus, 2023-03-21:
### Description

The run_lvs.py script in pdks/sky130A/libs.tech/klayout/lvs/run_lvs.py requires docopt, but it is not installed.

### Expected Behavior

Running run_lvs.py should work.

### Environment report

```shell
Kernel: Linux v6.0.12-76060012-generic
Distribution: pop 20.04
Python: v3.8.10 (OK)
Container Engine: docker v23.0.1 (OK)
OpenLane Git Version: 2857b632bb9a79f65c7ab6dca195d7eb5e5d60e6
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2857b632 2023-03-16T16:06:06+02:00 Fix Make Test Sets (#1683) - Mohamed Gaber -  (HEAD -> master, tag: 2023.03.17, origin/master, origin/HEAD)
735c148e 2023-03-16T00:18:12+02:00 Backport KLayout Script Updates (#1680) - Mohamed Gaber -  (tag: 2023.03.16)
05fac72e 2023-03-13T21:31:50+02:00 [BOT] Update openroad_app (#1654) - Openlane Bot -  (tag: 2023.03.14)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

git clone https://github.com/The-OpenROAD-Project/OpenLane.git
cd OpenLane
make mount
cd pdks/sky130A/libs.tech/klayout/lvs 
python3 run_lvs.py

### Relevant log output

```shell
Traceback (most recent call last):
  File "/openlane/pdks/sky130A/libs.tech/klayout/lvs/run_lvs.py", line 44, in <module>
    from docopt import docopt
ModuleNotFoundError: No module named 'docopt'
```

donn, 2023-03-22:
This script isn't part of OpenLane and is not presently used by any of its functions, so it not working is not exactly unexpected behavior.

It would be trivial to install docopt, however.

mguthaus, 2023-03-23:
The PCells in the PDKs directory also use an additional python package: numpy.

kareefardi, 2023-12-24:
Since this is not part of OpenLane, I believe there is no need for this. If you really need it, please reopen the issue.

closed 2023-12-24 by kareefardi