The-OpenROAD-Project/OpenLane #1691: segmentation fault in CTS

mattvenn, 2023-03-22:
### Description

hardening a small design used for a scanchain causes a segfault at CTS

### Expected Behavior

not to segfault

### Environment report

```shell
Kernel: Linux v5.15.0-67-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11039654/issue.tar.gz)

### Relevant log output

```shell
[STEP 10]                                                                    
[INFO]: Running Clock Tree Synthesis (log: ../home/matt/work/asic-workshop/shuttle9/tinytapeout-03/openlane/scanchain/runs/23_03_22_12_54/logs/cts/10-cts.
log)...                                                                                                                                                   
[INFO]: Executing openroad with Tcl script 'scripts/openroad/cts.tcl'...                                                                                  
OpenROAD 1f720d3b442e2cd8dc6c5372535320b18a105e8d                            
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.                                                         
[INFO]: Reading ODB at '/home/matt/work/asic-workshop/shuttle9/tinytapeout-03/openlane/scanchain/runs/23_03_22_12_54/results/placement/scanchain.odb'...
[INFO]: Setting RC values...                                                 
[INFO]: Configuring cts characterization...                                                                                                               
[INFO]: Performing clock tree synthesis...    
[INFO]: Looking for the following net(s):                                                                                                                 
[INFO]: Running Clock Tree Synthesis...                                      
[INFO CTS-0049] Characterization buffer is: sky130_fd_sc_hd__clkbuf_8.                                                                                    
[INFO CTS-0038] Number of created patterns = 50000.                                                                                                       
[INFO CTS-0038] Number of created patterns = 100000.                         
[INFO CTS-0039] Number of created patterns = 141984.                
[INFO CTS-0084] Compiling LUT.                                                                                                                            
Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
2           8           1           34          1           12          
[WARNING CTS-0043] 4896 wires are pure wire and no slew degradation.                                                                                      
TritonCTS forced slew degradation on these wires.                            
[INFO CTS-0046]     Number of wire segments: 37152.                                                                                                       
[INFO CTS-0047]     Number of keys in characterization LUT: 1488.
[INFO CTS-0048]     Actual min input cap: 1.
Signal 11 received                                                           
Stack trace:                                                                 
 0# 0x0000000000CD1057 in openroad                                           
 1# 0x00007FC21C19C400 in /lib64/libc.so.6                                  
 2# sta::Vertex::bfsInQueue(sta::BfsIndex) const in openroad                                                                                              
 3# sta::BfsIterator::enqueue(sta::Vertex*) in openroad                                                                                                   
 4# sta::ClkNetwork::findClkPins(bool, sta::Map<sta::Pin const*, sta::Set<sta::Clock*, std::less<sta::Clock*> >, std::less<sta::Pin const*> >&) in openroa
d
```

mattvenn, 2023-03-22:
resolved this by turning off CTS, but probably shouldn't do a segfault

maliberty, 2023-03-22:
While I will fix the segv it seems you are doing something wrong here.  You have buf_clk* inverters in your netlist already forming a clock tree.  Further they aren't legally placed:
[image]

maliberty, 2023-03-22:
You also have an sdc referring to an non-existent port
```create_generated_clock -name clk_scan_out -source [get_ports {clk_in}] -combinational -divide_by 1 [get_ports {clk_out}]```

there is no clk_out

maliberty, 2023-03-22:
It appears I was getting the wrong file loaded in the gui.  I'm still not sure what happened but you can ignore the above comments.

maliberty, 2023-03-22:
Though I do still see pre-existing clock buffers.

vijayank88, 2023-03-23:
@mattvenn 
Can you share your original `config.json`?

mattvenn, 2023-03-23:
https://github.com/TinyTapeout/tinytapeout-03/tree/main/openlane/scanchain

vijayank88, 2023-03-23:
@mattvenn 
https://github.com/TinyTapeout/tinytapeout-03/blob/222613bae52e65e4f14f28c48e4bd92a4e0764a6/openlane/scanchain/config.tcl#L44 here clock port assigned empty.
In `base.sdc` creating clock tree for `clk_in`.
So CTS stage unable to proceed.
Once `CLOCK_PORT` set empty, flow will proceed with `VIRTUAL_CLOCK` i think.
Can you update `config.tcl` for `CLOCK_PORT` and try?

closed 2023-03-23 by maliberty