The-OpenROAD-Project/OpenLane #1694: Integrate OpenRAM?

hanm2019, 2023-03-23:
### Description

The sram is commonly used in ASIC design. I found that the OpenLane doc give an example which use SRAM in `Designing a chip with an OpenRAM`. However, I found the OpenRAM is not integrate in the docker image, and I am confusion about how to use OpenRAM to generate the target sram, which is also be used in OpenLane.

### Proposal

It would be better to give a more detailed description in the doc, maybe integrate the OpenRAM will make the OpenLane easy to use.

armleo, 2023-03-24:
I am the author of the said documentation.

You can install Open PDKs with default settings using the commit that OpenLane wants. Then point PDK ROOT to that build of Open PDKs.

I don't know if the PDK installs the OpenRAMs by default or not. But it is not handled by Docker, but by Volare instead.

hanm2019, 2023-03-26:
Thanks, I will try

hanm2019, 2023-03-27:
I install OpenRAM and successed in generate the SRAM in 130nm, However. After I install OpenRAM with the following commands: 

```
cd $HOME/OpenRAM
make pdk

cd $HOME/OpenRAM
make install
```

and I get an error on OpenLane with `make test` :

```
(base) hanm@G1045:~/OpenLane$ make test
cd /home/hanm/OpenLane && \
	docker run --rm -v /home/hanm/OpenLane:/openlane -v /home/hanm/OpenLane/designs:/openlane/install -v /home/hanm/.volare:/home/hanm/.volare -e PDK_ROOT=/home/hanm/.volare -e PDK=sky130A  --user 1002:1002 -e DISPLAY=localhost:10.0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/hanm/.Xauthority:/.Xauthority --network host --security-opt seccomp=unconfined efabless/openlane:e5a7a0976918035d1a4312ac9965148f215aa4ed-amd64 sh -c "./flow.tcl -design spm -tag openlane_test -overwrite"
OpenLane e5a7a0976918035d1a4312ac9965148f215aa4ed
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/spm/config.json'...
[INFO]: PDK Root: /home/hanm/.volare
[INFO]: Process Design Kit: sky130A
no such variable
    (read trace on "::env(STD_CELL_LIBRARY)")
    invoked from within
"set ::env(TECH_LEF) "$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/techlef/$::env(STD_CELL_LIBRARY)__nom.tlef""
    (file "/home/hanm/.volare/sky130A/libs.tech/openlane/config.tcl" line 15)
    invoked from within
"source $pdk_config"
    (procedure "prep" line 170)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 411)
Makefile:183: recipe for target 'test' failed
make: *** [test] Error 1

```

armleo, 2023-03-27:
You dont need to rebuild the OpenRAM, Open PDKs comes with prebuilt SRAM macros, you just need to enable theur installatuon.

The volare version of OpenPDK that comes with openlane may or may not have the macro installation enabled.

hanm2019, 2023-03-27:
I remove the ~/.volare and reinstall the OpenLane. After that, I only run 
```
cd $HOME/OpenRAM
make install
```
for OpenRAM. but get
```
(openRAM) hanm@G1045:~/OpenRAM$ make install                                                                                                                                                                                         [3/180]
Cloning SRAM library...
正克隆到 '/home/hanm/.volare/sky130_fd_bd_sram'...
remote: Enumerating objects: 2659, done.
remote: Counting objects: 100% (2659/2659), done.
remote: Compressing objects: 100% (704/704), done.
remote: Total 2659 (delta 1990), reused 2600 (delta 1931), pack-reused 0
接收对象中: 100% (2659/2659), 1.12 MiB | 651.00 KiB/s, 完成.
处理 delta 中: 100% (1990/1990), 完成.
注意：正在检出 'a83b6468c48434d927b90058b22047843c58027b'。

您正处于分离头指针状态。您可以查看、做试验性的修改及提交，并且您可以通过另外
的检出分支操作丢弃在这个状态下所做的任何提交。

如果您想要通过创建分支来保留在此状态下所做的提交，您可以通过在检出命令添加
参数 -b 来实现（现在或稍后）。例如：

  git checkout -b <新分支名>

HEAD 目前位于 a83b646 Update colenda to have different device l/w.
Warning: /home/hanm/.volare/skywater-pdk not found!! Run make pdk first.
Makefile:86: recipe for target 'install' failed
make: *** [install] Error 1
```

It seem that I must run `make pdk` first

vijayank88, 2023-03-27:
@donn 
Is there a plan to integrate OpenRAM part of OpenLane docker image as per @hanm2019 requirement?

donn, 2023-03-27:
@vijayank88 Nope.

closed 2023-04-13 by donn