The-OpenROAD-Project/OpenLane #1697: Synthesis results can change based on comments or blank lines in Verilog source

algofoogle, 2023-03-25:
### Description

Depending on how many blank lines (or comment-only lines) I have within the first ~95 lines of my Verilog source, I get different synthesis results (and hence different routing/layout) when running (say) `make MYTEST` to harden my design. There is some wiggle room: 20 blank lines and 30 blank lines give identical synthesis results to each other, but 0 and 40 blank lines do not.

This doesn't happen with all source files. I've provided a small example below where it DOES happen: I've got one source file with my top design, and adding blank lines to this makes no difference at all. However, the top design instantiates a module that is defined in another file, and THAT file exhibits this strange behaviour when blank lines are added to it.

I think this is a Yosys issue, i.e. it starts with synthesis and straight away I can detect a change in `results/synthesis/MYTEST.v` after Step 1 is complete.

(Tested using [efabless/caravel_user_project](https://github.com/efabless/caravel_user_project) with both `mpw-9a` and `mpw-8c` tags; results vary a tiny bit between tags, but are consistent within the same tag each time).

### Expected Behavior

No matter how many blank lines or comment-only lines there are scattered throughout all included `.v` source files, so long as the material *code* itself is not changed, then synthesis should remain the same every time if all other variables/parameters remain the same.

This stood out as a problem because I didn't change anything else about my already-hardened design, other than to add the SPDX/license header comments to all my files, and then I found this lead to unexpected differences in my final GDS -- which I noticed first because the `routing_layer5_pct` (and others) layer congestion percentage had changed when doing a final-FINAL test.

### Environment report

```shell
Kernel: Linux v5.15.0-67-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

I cloned [efabless/caravel_user_project](https://github.com/efabless/caravel_user_project):
```bash
git clone git@github.com:efabless/caravel_user_project
git checkout mpw-9a
```

Then dropped in contents of [MYTEST.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11068101/MYTEST.tar.gz):
```
verilog/rtl/MYTEST.v          # Top design; instantiates MYTEST_sub.
verilog/rtl/MYTEST_sub.v      # The file where blank lines change synthesis results!!
openlane/MYTEST/config.json   # Uses the above 2 Verilog files.
```

I ran: `make MYTEST`

I then added 20 blank lines to the top of `MYTEST_sub.v` and ran the make again -- got **different** synthesis results.

I added 10 more blank lines (30 total), and ran the make -- got **SAME** synthesis results as for 20 blank lines.

I added 10 more blank lines (40 total), and ran the make -- got **DIFFERENT** synthesis results again.

### Relevant log output

```shell
I don't yet know if there's any relevant log output because what we're looking at is a difference between runs; not the outcome of a run in isolation. I will try to add more if I find anything telling, however.
```

algofoogle, 2023-03-25:
My current theory (all speculation but based on sniffing around `logs/synthesis/1-synthesis.log`): Yosys creates internal names for registers (etc) that are based on the line and column number of where that identifier is defined in the source. These go into some internal master list and then get sorted just as simple strings. As strings, `99` comes after `98`, but `100` comes *before* `99`.

It'd be no surprise then if certain things near a magnitude change of (say) 2 digits to 3 (99 &rarr; 100) would end up being sorted differently if line numbers are pushed around, and hence some things would be prioritised differently during synthesis.

vijayank88, 2023-03-25:
@algofoogle 
Can you attach the `resutls/synthesis/MYTEST.v`? so that it is easy to compare.

algofoogle, 2023-03-25:
Here are those files resulting from 0 leading blank lines, then 20, 30, and 40:
[MYTEST_synth.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11068478/MYTEST_synth.tar.gz)

I'll make a script to run a bunch of tests with different numbers of blank lines at the start, and then package up the `results/synthesis/MYTEST.v` files from each. I'm sort of finding that I'm getting the same results as for 920 blank lines, 930, and 940, which kinda supports my last comment.

vijayank88, 2023-03-25:
Yes, I can reproduce the same.
20-30 blank lines produces same netlist.
0-20 have the differences.
As it is related to Yosys, I leave this to OpenLane team to respond it.
Thanks for reporting it.

vijayank88, 2023-03-25:
@algofoogle 
How you're adding empty lines in verilog file?
Is just new lines or using `//` character as per verilog standard to add commented lines?

algofoogle, 2023-03-25:
They both seem to have the same result. Early on my testing was with `//` comment lines, but I was getting exactly the same results just from blank lines, so more recently I've only been testing that way.

vijayank88, 2023-03-25:
Can you report this to Yosys directly? https://github.com/YosysHQ/yosys/issues

algofoogle, 2023-03-25:
I can do that but I might have trouble describing how to reproduce the issue because I'm not sure how to separate it from what OpenLane is doing specifically. Does OpenLane (or the caravel make process) log exact commands it uses to run yosys (so I can just copy that as part of the bug report), or would referring back to this issue provide enough context for them to work it out and reproduce it?

algofoogle, 2023-03-28:
Yeah, I think it makes sense to just let OpenLane pass that responsibility to whatever does its synthesis (which for now is Yosys). There *could* be a `NORMALIZE_SOURCE_WHITESPACE` config option or whatever (in this case "whitespace" referring to actual whitespace but also comment tokens) but that would cause other mess and confusion I'm sure.

maliberty, 2023-03-28:
Is there an associated yosys discussion?

oharboe, 2024-07-04:
@maliberty I'm thinking this might not only be a problem, it could also be a solution. I see significant changes in quality of results for what should be a no-op change in input to Yosys: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2101

The oportunity is that changing the "seed" for Yosys synthesis could lead to better quality of results. In lieu of a supported seed feature in Yosys, one could insert blank lines randomly in the input Verilog code and re-run the OpenROAD flow and pick the best result. EDA tools are know to have very significant differences in quality of results for seemingly innocuous changes in the input.

donn, 2024-07-04:
Oh right this is still open

closed 2024-07-04 by donn

oharboe, 2024-07-04:
@donn Did you intend to close this issue?

I just shared a non-sequitor that fuzzing input to Yosys could be a useful feature as evidenced by the change in quality of results seen when making what should be no-op changes to Yosys input...

dlmiles, 2024-07-04:
+1 to keeping this feature as-is.  This is most useful as-is.

Maybe OL2 can provide an example of a normalizer/fuzzer/whatever plugin(s) that demonstrate how to mutate the necessary things at appropriate steps for those scenarios that are looking for:
 * white space to not affect results (maybe [verible-verilog-format](https://chipsalliance.github.io/verible/verilog_format.html) module here)
 * inactive verilog code to not affect results (maybe front-end pre-processor pass, maybe it needs to insert blank lines for removed ifndef section to keep line number the same)
 * injecting fuzzing/renaming to achieve different results
 
Now everyone gets what they want out of the system and it is documented how and where to put a plugin to achieve what you want from the flow.

oharboe, 2024-07-04:
@dlmiles I'm using Yosys with Bazel and I want to canonincalize the Verilog input for that reason(can save hours in build times by skipping synthesis for no-op changes).

Fortunately, Yosys supports canonicalization of input by first reading in the Verilog, then using using write_json, then restarting Yosys and reading in the canonicalized input using read_json.

I have proposed such canonicalization for OpenROAD-flow-scripts in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2104

Ravenslofty, 2024-07-04:
As a Yosys developer, I would suggest using RTLIL format instead of JSON format, if you're going to do that.

oharboe, 2024-07-04:
@Ravenslofty I also want to clean out unused modules. Can I do that with running "proc"?

Ravenslofty, 2024-07-04:
Use `opt_clean -purge` for that.

oharboe, 2024-07-04:
@Ravenslofty Like so?

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2104/commits/725fad6924b084e941f73e783f5af3466001219d#diff-8138cdd88d0792f13d7fbe17a3532f0601186c9224ba1d5fba892eaedb55b297

Ravenslofty, 2024-07-04:
That looks good to me, as long as `proc` is called when doing synthesis after canonicalisation; `proc` and `hierarchy` should be the first two commands in any synthesis flow, because they perform elaboration of Verilog, and things go very badly wrong when you forget them (which is why many commands have a "call proc first" error).

oharboe, 2024-07-04:
> That looks good to me, as long as `proc` is called when doing synthesis after canonicalisation; `proc` and `hierarchy` should be the first two commands in any synthesis flow, because they perform [...]

`synth -top -run :fine` is being run first, so that should be in order. According to the documentation, https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth.html, `hierarchy -check [-top <top> | -auto-top]`, then `proc` are run first.

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/3629b63102f12b9a113a6ee6da6ae0cb3a9c8679/flow/scripts/synth_preamble.tcl#L104

maliberty, 2024-07-04:
@oharboe "seed sweeping" is a well know trick and applies to many algorithms.  Its quite popular in FPGAs

oharboe, 2024-07-04:
> @oharboe "seed sweeping" is a well know trick and applies to many algorithms. Its quite popular in FPGAs

Yes, I am familiar. I used it in a project to eke it over the finish line.... 

I have also worked on stocastic geoscience software. Some customers wanted seeds to be compatible across versions, which was of course impossible. It would have to be the exact same version and same bugs and all to work exactly the same :-)