The-OpenROAD-Project/OpenLane #1699: FP_IO_MIN_DISTANCE does not affect the capacity of pin in placement

hanm2019, 2023-03-28:
### Description

I am trying to create a MACRO using OpenLane. 

This macro is a part of the wrapper design, and will be reused many time in the wrapper module. The input and output signals is huge (about 9000+), so that I get the error information in STEP 4
```
[INFO]: Running IO Placement (log: designs/CellSARA/runs/30Percent_IO/logs/floorplan/4-place_io.log)...
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/io_place.py --config /openlane/designs/CellSARA/pin_order.cfg --hor-layer met3 --ver-layer met2 --ver-width-mult 2 --hor-width-mult 2 --hor-extension 0
--ver-extension 0 --length 4 --unmatched-error --input-lef /openlane/designs/CellSARA/runs/30Percent_IO/tmp/merged.nom.lef --output-def /openlane/designs/CellSARA/runs/30Percent_IO/tmp/floorplan/4-io.def --output /openlane/designs/CellS
ARA/runs/30Percent_IO/tmp/floorplan/4-io.odb /openlane/designs/CellSARA/runs/30Percent_IO/tmp/floorplan/3-initial_fp.odb |& tee /dev/null /openlane/designs/CellSARA/runs/30Percent_IO/logs/floorplan/4-place_io.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
OpenROAD 1a1617d908d2ebdb731de9ab4e3d9fd93a6dcf97
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Top-level design name: CellSARA
Block boundaries: 0 0 954980 965700
Horizontal Tracks Origin: 340, Count: 1420, Step: 680
Vertical Tracks Origin: 230, Count: 2076, Step: 460
There are more pins/virtual_pins: 2542, than places to put them: 2076. Try making your floorplan area larger.
child process exited abnormally
```
It appears that the floorplan area is insufficient to accommodate all pins, and I have observed some gaps between pins in the `SPM` design. Therefore, I suppose that adjusting the distance between pins might enable the floorplan to accommodate more pins.
   
I found that there is a parameter `FP_IO_MIN_DISTANCE`, which the default value is 3. so I set it to 0.2 in `config.json`.

the config.json are:
```
{
    "DESIGN_NAME": "CellSARA",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 10.0,
    "DESIGN_IS_CORE": false,
    "FP_PDN_CORE_RING": false,
    "FP_IO_MIN_DISTANCE": 0.20,
    "FP_CORE_UTIL": 30,
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg"
}
```

However, even after re-running the flow, I still encounter the same error message. It appears that changing `FP_IO_MIN_DISTANCE` does not impact the pin placement capacity.

### Expected Behavior

By reducing the FP_IO_MIN_DISTANCE value, the pin placement capacity will increase.

### Environment report

```shell
Kernel: Linux v5.4.0-87-generic
Distribution: ubuntu 18.04
Python: v3.8.3 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: e5a7a0976918035d1a4312ac9965148f215aa4ed
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e5a7a09 2023-03-22T15:29:01+02:00 Rework Placement (#1684) - Mohamed Gaber -  (grafted, HEAD -> master, tag: 2023.03.23, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[CellSARA.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11085271/CellSARA.zip)

### Relevant log output

```shell
OpenLane Container (e5a7a09):/openlane$ ./flow.tcl -design CellSARA -tag 30Percent_IO -overwrite
OpenLane e5a7a0976918035d1a4312ac9965148f215aa4ed
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/CellSARA/config.json'...
[INFO]: PDK Root: /home/hanm/.volare
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/CellSARA/runs/30Percent_IO
[INFO]: Removing existing /openlane/designs/CellSARA/runs/30Percent_IO...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[STEP 1]                                                                                                                                                                                                                           [249/376]
[INFO]: Running Synthesis (log: designs/CellSARA/runs/30Percent_IO/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/CellSARA/runs/30Percent_IO/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: designs/CellSARA/runs/30Percent_IO/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 943.92 and height 943.84.
[STEP 4]
[INFO]: Running IO Placement (log: designs/CellSARA/runs/30Percent_IO/logs/floorplan/4-place_io.log)...
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/io_place.py --config /openlane/designs/CellSARA/pin_order.cfg --hor-layer met3 --ver-layer met2 --ver-width-mult 2 --hor-width-mult 2 --hor-extension 0
--ver-extension 0 --length 4 --unmatched-error --input-lef /openlane/designs/CellSARA/runs/30Percent_IO/tmp/merged.nom.lef --output-def /openlane/designs/CellSARA/runs/30Percent_IO/tmp/floorplan/4-io.def --output /openlane/designs/CellS
ARA/runs/30Percent_IO/tmp/floorplan/4-io.odb /openlane/designs/CellSARA/runs/30Percent_IO/tmp/floorplan/3-initial_fp.odb |& tee /dev/null /openlane/designs/CellSARA/runs/30Percent_IO/logs/floorplan/4-place_io.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
OpenROAD 1a1617d908d2ebdb731de9ab4e3d9fd93a6dcf97
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Top-level design name: CellSARA
Block boundaries: 0 0 954980 965700
Horizontal Tracks Origin: 340, Count: 1420, Step: 680
Vertical Tracks Origin: 230, Count: 2076, Step: 460
There are more pins/virtual_pins: 2542, than places to put them: 2076. Try making your floorplan area larger.
child process exited abnormally

[INFO]: Saving current set of views in 'designs/CellSARA/runs/30Percent_IO/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/CellSARA/runs/30Percent_IO/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/CellSARA/runs/30Percent_IO/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2023-03-28:
@hanm2019 
your die size as per floorplan report `0.0 0.0 954.98 965.7`.
You can't increase further?

hanm2019, 2023-03-28:
Of course, if I increase the die size, it can place more pins under the min distance between pin is 3. But this design is very sample, and there are not much nets in it. Increasing the die size will turn the low utilization of die. So that, It maybe 
more economical to decrease the distance between the pin.

vijayank88, 2023-03-28:
Can you share the `src` files and `config.json`?

hanm2019, 2023-03-28:
Yes, I have upload the design/CellSARA in `Reproduction material` subseciton

vijayank88, 2023-03-28:
@maliberty 
Is automated die size calculation include number of IO pins or not?
What is the recommended way here?

vijayank88, 2023-03-28:
@hanm2019 
If I remove `FP_PIN_ORDER_CFG` and with fixed die size of `1500um` pin placement passing successfully. 
For fixed pin location, you should provide right die size to pass the flow.

hanm2019, 2023-03-28:
https://drive.google.com/file/d/12C_2CaXRM5imwYcrz-_DPbWD7RwL9Pot/view?usp=share_link
https://drive.google.com/file/d/1vhun8OxrMSQLtF9sKL6NbD3BLHwFGqW9/view?usp=share_link

Due to the zip file is large than 25M, I upload it on google. Can you download it?

vijayank88, 2023-03-28:
Missing file: 
```
[ERROR STA-0511] cannot open 'openlane/scripts/openroad/resizer_routing_design.tcl'.
Error: run.tcl, 57 STA-0511
```
Can compress and upload as single file?

Can update `"PL_TARGET_DENSITY":` by lowering the density from default value and try?

hanm2019, 2023-03-28:
https://drive.google.com/file/d/1f4SRqtKSTvc2u181VYOtBpXOyyxhdo0T/view?usp=share_link 
here is the single file

hanm2019, 2023-03-28:
> Missing file:
> 
> ```
> [ERROR STA-0511] cannot open 'openlane/scripts/openroad/resizer_routing_design.tcl'.
> Error: run.tcl, 57 STA-0511
> ```
> 
> Can you compress and upload as single file?
> 
> Can you update `"PL_TARGET_DENSITY":` by lowering the density from default value and try?

what value should I set to `PL_TARGET_DENSITY` ?

vijayank88, 2023-03-28:
`PL_TARGET_DENSITY: 0.40` try that.

```
#S
$2
io_out_r_data_0_.*
```
What is `$2` meant in `pin_order.cfg`?

hanm2019, 2023-03-28:
I try `PL_TARGET_DENSITY: 0.40` and get 
```
[STEP 12]
[INFO]: Running Global Routing Resizer Design Optimizations (log: designs/CellSARA/runs/1500/logs/routing/12-resizer_design.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: designs/CellSARA/runs/1500/logs/routing/12-resizer_design.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 170774
[INFO GRT-0198] Via related Steiner nodes: 3419
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 344118
[INFO GRT-0112] Final usage 3D: 2448295
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_design.tcl, 39 GRT-0118
child process exited abnormally

[ERROR]: Creating issue reproducible...
```
the zip file is here:
https://drive.google.com/file/d/1EuE7Kd3Uj7zIChBbpfRckJwCG1hOlU7z/view?usp=share_link

hanm2019, 2023-03-28:
> `PL_TARGET_DENSITY: 0.40` try that.
> 
> ```
> #S
> $2
> io_out_r_data_0_.*
> ```
> 
> What is `$2` meant in `pin_order.cfg`?

the $2 mean there are two virtual pins

vijayank88, 2023-03-29:
@kareefardi 
Thanks for the update.
I also just measured the distance between the pins and it shows `0.08um`, but actually i passed `1um`.
Is there any example to increase pin distance if we pass with `pin_order.cfg`?

hanm2019, 2023-03-29:
> @hanm2019 Do you want to place the pins as per fixed `pin_order.cfg`?

Yes, I want to place the pin in order and also in a small distance

hanm2019, 2023-03-29:
I have increase the die size to 1600x1600, and it can place all pins, but I met another error in STEP 12. https://github.com/The-OpenROAD-Project/OpenLane/issues/1699#issuecomment-1486888047

It seen like you have succeeded in generating the GDSII for the CellSARA? could you share me your config file :>

maliberty, 2023-03-29:
0.08 is less than min space (0.3) so this is already illegal.

maliberty, 2023-03-29:
Your extremely dense IOs coupled with fair amount of hold buffering is going to make routing very difficult.  For example:
[image]
(input buffers in yellow, hold buffers in green).

This block looks to be very IO limited and needs more perimeter or you need to shift pins to multiple layers.

hanm2019, 2023-03-30:
How can I shift pins to multiple layers?

vijayank88, 2023-03-30:
@hanm2019 
Check the following [README.md](https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/ppl) to understand how pins placed with different layers.
But how to implement with OpenLane need to check with OpenLane team.

hanm2019, 2023-04-03:
Let me introduce my architecture design to determine the configuration is suit for it. 

I would like to design a 2D PE array like  chip. The PE-Array architecture is here:
[image]
It has 16x16 PE, and each PE is connect with the neighbor PEs. 

Since this PE-Array is big but easy structure, I split it into a lot of 4x4 PE sub-array. and then use OpenLane to generate a 4x4 PE sub-array's GDSII file. Finally, I use 16 GDSII of sub-arrays to join for the 16x16 PE array.

To make the splicing easy, I use `pin_order.cfg` file to place the pins, and `DESIGN_IS_CORE` is set to `false`

the config file is 
```
{
    "DESIGN_NAME": "CellSARA",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 10.0,
    "DESIGN_IS_CORE": false,
    "FP_PDN_CORE_RING": false,
    "FP_IO_MIN_DISTANCE": 0.20,
    "FP_CORE_UTIL": 30,
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg"
}
```

vijayank88, 2023-04-04:
@hanm2019 
Follow up this #1711 enhancement request. Hope it will resolve your issue.

hanm2019, 2023-04-04:
I have an another question that : is the pin wider than the line in the die? 
for example: There are a module has 100 input pins and 100 output pins, and the logic is very sample

```
module A(
input clk,
input [99:0] a,
output reg [99:0] b) 

always @(posedge clk) begin
 b <= a;
end

endmodule
```

I want to place the `a` pins at the west of die while the `b` pins at the east of die

Does the line in the die can be more closer than on the pin? such as 

[image]

kareefardi, 2023-04-10:
@hanm2019 So at the moment you can set min_distance for a specific direction using the config file. See FP_PIN_ORDER_CFG in https://openlane.readthedocs.io/en/latest/reference/configuration.html. A pin follows that same spacing and width rules of it's metal layer as a wire. In your current configuration you can set FP_IO_HTHICKNESS_MULT and FP_IO_VTHICKNESS_MULT to one, in order to guarantee maximum placement of pins. This might result in more congestion but this is the way to maximize number of pins per side.

hanm2019, 2023-04-10:
Thanks! I will have a try. By the way, How can I achieve the same effection of `FP_IO_HTHICKNESS_MULT` in ORFS?

kareefardi, 2023-06-26:
This has been addressed in OpenLane. Please open a new issue if the issue still persists.

closed 2023-06-26 by kareefardi