The-OpenROAD-Project/OpenLane #1707: Installation Error

Prashanthrajagopal91, 2023-03-29:
Hello,

I am trying to install openlane with the command line prompt, I am facing this error while using "sudo make" 

cd $HOME
git clone https://github.com/The-OpenROAD-Project/OpenLane
cd OpenLane
**make**
make test

The error is:

The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/prashanth/OpenLane/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

Makefile:145: recipe for target 'venv/created' failed
make: *** [venv/created] Error 1

Can anyone please help understand this better and sort out? Thanks!

donn, 2023-03-29:
Please *accurately* follow the setup guide. You need to install a couple packages on Ubuntu.

https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html

closed 2023-03-29 by donn

Prashanthrajagopal91, 2023-03-29:
Thank you so much! it worked up to the basic test.

But now I am facing a different error with the klayout. Can you please help me understand this error?

OpenLane Container (ed19423):/openlane$ klayout -e -nn $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyt    -l $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyp    ./designs/spm/runs/openlane_test/results/final/gds/spm.gd
process 36: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
^C
OpenLane Container (ed19423):/openlane$ klayout -e -nn $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyt
process 41: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
^C
OpenLane Container (ed19423):/openlane$ klayout
process 46: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace