The-OpenROAD-Project/OpenLane #1710: Crash in CTS when building design that outputs inverted clock

MichaelBell, 2023-03-30:
### Description

Building a very simple Tiny Tapeout project which simply inverts io_in[0] and outputs it on io_out[0], the CTS crashes.  

io_in[0] is configured as the clock for Tiny Tapeout.  

The repro is available here: https://github.com/MichaelBell/tt03-invert-clock

### Expected Behavior

The project should build successfully

### Environment report

```shell
Kernel: Linux v5.10.60.1-microsoft-standard-WSL2
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11114968/issue_reproducible.tar.gz)

### Relevant log output

```shell
Setting global connections for newly added cells...
Writing OpenROAD database to /work/runs/wokwi/results/cts/repro_top.odb...
Writing layout to /work/runs/wokwi/results/cts/repro_top.def...
Writing timing constraints to /work/runs/wokwi/results/cts/repro_top.sdc...
Signal 11 received
Stack trace:
 0# 0x0000000000CD1057 in openroad
 1# 0x00007F76CAC45400 in /lib64/libc.so.6
 2# sta::ConcreteNetwork::name(sta::Port const*) const in openroad
 3# sta::SdcNetwork::portName(sta::Pin const*) const in openroad
 4# sta::Network::pathNameCmp(sta::Pin const*, sta::Pin const*) const in openroad
 5# sta::PortDelayLess::operator()(sta::PortDelay const*, sta::PortDelay const*) const in openroad
 6# void std::__insertion_sort<__gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, __gnu_cxx::__ops::_Iter_comp_iter<sta::PortDelayLess> >(__gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, __gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, __gnu_cxx::__ops::_Iter_comp_iter<sta::PortDelayLess>) in openroad
 7# void std::__merge_sort_with_buffer<__gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, sta::PortDelay**, __gnu_cxx::__ops::_Iter_comp_iter<sta::PortDelayLess> >(__gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, __gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, sta::PortDelay**, __gnu_cxx::__ops::_Iter_comp_iter<sta::PortDelayLess>) in openroad
 8# void std::__stable_sort_adaptive<__gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, sta::PortDelay**, long, __gnu_cxx::__ops::_Iter_comp_iter<sta::PortDelayLess> >(__gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, __gnu_cxx::__normal_iterator<sta::PortDelay**, std::vector<sta::PortDelay*, std::allocator<sta::PortDelay*> > >, sta::PortDelay**, long, __gnu_cxx::__ops::_Iter_comp_iter<sta::PortDelayLess>) in openroad
 9# sta::WriteSdc::writeOutputDelays() const in openroad
10# sta::WriteSdc::writeTiming() const in openroad
11# sta::writeSdc(sta::Instance*, char const*, char const*, bool, bool, int, bool, bool, sta::Sdc*) in openroad
12# sta::Sta::writeSdc(char const*, bool, bool, int, bool, bool) in openroad
13# 0x0000000000E99A47 in openroad
14# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
15# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
16# TclObjInterpProcCore in /lib64/libtcl8.5.so
17# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
18# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
19# TclObjInterpProcCore in /lib64/libtcl8.5.so
20# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
21# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
22# 0x00007F76CF7C7647 in /lib64/libtcl8.5.so
23# TclEvalObjEx in /lib64/libtcl8.5.so
24# 0x00007F76CF80227F in /lib64/libtcl8.5.so
25# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
26# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
27# 0x00007F76CF7C7647 in /lib64/libtcl8.5.so
28# TclEvalObjEx in /lib64/libtcl8.5.so
29# 0x00007F76CF7821D0 in /lib64/libtcl8.5.so
30# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
31# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
32# 0x00007F76CF7C7647 in /lib64/libtcl8.5.so
33# TclEvalObjEx in /lib64/libtcl8.5.so
34# 0x00007F76CF785F00 in /lib64/libtcl8.5.so
35# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
36# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
37# TclObjInterpProcCore in /lib64/libtcl8.5.so
38# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
39# 0x00007F76CF7BF36C in /lib64/libtcl8.5.so
40# TclObjInterpProcCore in /lib64/libtcl8.5.so
41# 0x00007F76CF77AEB2 in /lib64/libtcl8.5.so
42# 0x00007F76CF77BF1E in /lib64/libtcl8.5.so
43# Tcl_EvalEx in /lib64/libtcl8.5.so
44# Tcl_Eval in /lib64/libtcl8.5.so
45# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
46# ord::tclAppInit(Tcl_Interp*) in openroad
47# Tcl_Main in /lib64/libtcl8.5.so
48# main in openroad
49# __libc_start_main in /lib64/libc.so.6
50# 0x0000000000CCB7E7 in openroad
```

maliberty, 2023-03-30:
There was a recent issue in OR that looks similar.  You might try the head of master in OR to see if it solves your problem.

MichaelBell, 2023-03-31:
I still see a crash in the same place using the 2023.03.30 OpenLane tag: https://github.com/MichaelBell/tt03-invert-clock/actions/runs/4579228600/jobs/8086766240

I'm not sure how to set up OpenROAD independently.

maliberty, 2023-04-01:
You can look at updating the OR version in https://github.com/The-OpenROAD-Project/OpenLane/tree/master/docker#updating-a-tool-binary

@donn can you help here?

donn, 2023-04-01:
I'll update it tomorrow.

vijayank88, 2023-04-01:
I confirm the latest OR resolves the issues.

kareefardi, 2023-04-09:
Blocked on https://github.com/The-OpenROAD-Project/OpenROAD/issues/3131

closed 2023-04-18 by kareefardi