The-OpenROAD-Project/OpenLane #1711: Provide option to increase pin spacing with pin_order.cfg

vijayank88, 2023-04-01:
### Description

Based on comment https://github.com/The-OpenROAD-Project/OpenLane/issues/1699#issuecomment-1488316946
For IO limited block with fixed pin location through `pin_order.cfg` it is good to add `FP_IO_MIN_DISTANCE` configuration to keep some minimum distance.

[image]
in above floorplan, the corner has some more space. As existing pin placement not increasing pin distance. It is good to customize pin distance for better pin placement.

So it will be easy to place buffer and routing, otherwise facing congestion issue.

kareefardi, 2023-04-03:
@vijayank88 what is the exact configuration that you used to take that screenshot?

vijayank88, 2023-04-03:
Increased die size 1600x3010 and ran the OpenLane flow with user design of fixed pin_order.cfg.

vijayank88, 2023-04-04:
Yes, I agree. Die size should be increased to place all the pins...

hanm2019, 2023-04-04:
make be place in multiple layer? Can I decide the pin's layer also in pin_order.cfg? such as
```
#LAYER 1
#N 
clk
rst
bypass

#S
xxxx

#LAYER 2
#N 
...
...

#S
...
...

```