The-OpenROAD-Project/OpenLane #1713: Getting this error when running a 8 bit ripple carry adder flow

S-PADMANABAN-2001, 2023-04-03:
### Description

While running `./flow.tcl -design rca` got the below error. Failing in step 6 PDN error and floorplan failed.

### Expected Behavior

flow run and finish successfully and all the reports should me generated

### Environment report

```shell
<pre><font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ git --version
git version 2.17.1
<font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ docker --version
Docker version 23.0.2, build 569dd73
<font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ python3 --version
Python 3.6.9
<font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ python3 -m pip --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)
<font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later &lt;http://gnu.org/licenses/gpl.html&gt;
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
<font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ python3 -m venv -h
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear]
            [--upgrade] [--without-pip] [--prompt PROMPT]
            ENV_DIR [ENV_DIR ...]

Creates virtual Python environments in one or more target directories.

positional arguments:
  ENV_DIR               A directory to create the environment in.

optional arguments:
  -h, --help            show this help message and exit
  --system-site-packages
                        Give the virtual environment access to the system
                        site-packages dir.
  --symlinks            Try to use symlinks rather than copies, when symlinks
                        are not the default for the platform.
  --copies              Try to use copies rather than symlinks, even when
                        symlinks are the default for the platform.
  --clear               Delete the contents of the environment directory if it
                        already exists, before environment creation.
  --upgrade             Upgrade the environment directory to use this version
                        of Python, assuming Python has been upgraded in-place.
  --without-pip         Skips installing or upgrading pip in the virtual
                        environment (pip is bootstrapped by default)
  --prompt PROMPT       Provides an alternative prompt prefix for this
                        environment.

Once an environment has been created, you may wish to activate it, e.g. by
sourcing an activate script in its bin directory.
<font color="#8AE234"><b>padmanaban@padmanaban-VirtualBox</b></font>:<font color="#729FCF"><b>~/OpenLane</b></font>$ 
</pre>
```

### Reproduction material

config.json
{
    "DESIGN_NAME": "rca",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 10.0,
    "DESIGN_IS_CORE": true
}
[... 77 lines cut ...]
<font color="#CC0000">    (procedure &quot;run_tcl_script&quot; line 211)</font>
<font color="#CC0000">    invoked from within</font>
<font color="#CC0000">&quot;run_tcl_script -tool openroad -no_consume {*}$args&quot;</font>
<font color="#CC0000">    (procedure &quot;run_openroad_script&quot; line 2)</font>
<font color="#CC0000">    invoked from within</font>
<font color="#CC0000">&quot;run_openroad_script $::env(SCRIPTS_DIR)/openroad/pdn.tcl  -indexed_log [index_file $::env(floorplan_logs)/pdn.log]  -save &quot;to=$::env(floorplan_tmpfile...&quot;</font>
<font color="#CC0000">    (procedure &quot;gen_pdn&quot; line 15)</font>
<font color="#CC0000">    invoked from within</font>
<font color="#CC0000">&quot;gen_pdn&quot;</font>
<font color="#CC0000">    (procedure &quot;run_power_grid_generation&quot; line 65)</font>
<font color="#CC0000">    invoked from within</font>
<font color="#CC0000">&quot;run_power_grid_generation&quot;</font>
<font color="#CC0000">    (procedure &quot;run_floorplan&quot; line 56)</font>
<font color="#CC0000">    invoked from within</font>
<font color="#CC0000">&quot;run_floorplan&quot;} -errorline 1</font>
<font color="#06989A">[INFO]: Saving current set of views in &apos;designs/rca/runs/RUN_2023.04.03_12.15.41/results/final&apos;...</font>
<font color="#06989A">[INFO]: Generating final set of reports...</font>
<font color="#06989A">[INFO]: Created manufacturability report at &apos;designs/rca/runs/RUN_2023.04.03_12.15.41/reports/manufacturability.rpt&apos;.</font>
<font color="#06989A">[INFO]: Created metrics report at &apos;designs/rca/runs/RUN_2023.04.03_12.15.41/reports/metrics.csv&apos;.</font>
<font color="#06989A">[INFO]: Saving runtime environment...</font>
<font color="#CC0000">[ERROR]: Flow failed.</font>
<font color="#06989A">[INFO]: The failure may have been because of the following warnings:</font>
[WARNING]: Current core area is too small for the power grid settings chosen. The power grid will be scaled down.
</pre>
```

d-m-bailey, 2023-04-03:
Looks like the resulting floorplan is too small for power routing.
```
[WARNING]: Current core area is too small for the power grid settings chosen. The power grid will be scaled down.
```
Try fixing the core size to something bigger in the config file. Maybe 200 x 200.

donn, 2023-04-13:
Staled out.