The-OpenROAD-Project/OpenLane #1718: Document how to debug and fix programs

xobs, 2023-04-05:
### Description

Sometimes, program in the openlane flow segfault and need to be fixed. However, it is unclear how to attach a debugger to these programs, or how they work with the rest of the system.

It would be nice to document how to attach a debugger to programs such as openroad when run under the openlane scripts.

xobs, 2023-04-05:
How do I use the reproducibles? It's not clear how to access the programs within the Docker container, since it's managed by the Makefile.

mithro, 2023-04-05:
If a user wants to debug a problem using gdb themselves, how would you do it?

donn, 2023-04-05:
> How do I use the reproducibles? It's not clear how to access the programs within the Docker container, since it's managed by the Makefile.

The Makefile and the Docker container both are no longer relevant at this point- they exist only as a convenience for running pre-compiled tools for the convenience of users. The moment debugging is a consideration, you are going to need to compile and install OpenROAD (or any other tools) on your own machine by adding it to `PATH`.

---
You can then simply run the tool with `run.tcl` with your debugger as you would any other program- I personally use lldb, so here's how it'd work:

```sh
cd designs/spm/runs/RUN_2023.04.05_23.41.08/issue_reproducible
lldb openroad -- ./run.tcl
> run
```

It's not particularly complex.

donn, 2023-04-10:
May I assume this has been addressed adequately at this juncture, @mithro?

ShvetankPrakash, 2023-07-13:
Hi @donn I am trying the solution you presented for debugging! I have installed OpenROAD and lldb natively. However when I follow your steps I get the following issue:
```
$ pwd
~/OpenLane/designs/spm/runs/RUN_2023.07.13_10.43.23/issue_reproducible
$ lldb openroad -- ./run.tcl 
Current executable set to 'openroad' (x86_64).
(lldb) run
error: process launch failed: Child exec failed.
(lldb) 
```
Is `run` the correct command? Please let me know if you have any guidance for how to debug. I am getting a segfault at the moment so I tried this setup I found here in this past issue. 

Thanks!