The-OpenROAD-Project/OpenLane #1724: No clock nets have been found during clock tree synthesis, but net found for clock?

jackpanderson, 2023-04-07:
### Description

[12-cts.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11182019/12-cts.log)
[15-resizer_design.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11182047/15-resizer_design.log)

I have been trying to push a design through the OpenLane toolflow and have been running into a recurring error with the clock net. Synthesizing any of the default designs runs as expected, but our design runs into the same issue. 

### Expected Behavior

It is expected that the CLK net is found for the clock, it seems like it is not being properly detected.

### Environment report

```shell
Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: bff79871616e51a61f0cbefdbdfb7467c749aa03
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: e6f9c8876da77220403014b116761b0b2d79aab4, tested: 12df12e2e74145e31c5a13de02f9a1e176b56e67)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

bff79871 2023-04-07T15:22:59-07:00 Merge branch 'The-OpenROAD-Project:master' into master - jackpanderson -  (HEAD -> master, origin/master, origin/HEAD)
587b76f6 2023-04-07T08:47:03-07:00 backup - jackpanderson -  ()
cb634fd5 2023-04-06T13:41:52+02:00 Fix ci go package install (#1722) - Kareem Farid -  ()
---
Git Remotes

origin  https://github.com/Cal-Poly-RAMP/CARPOpenLane (fetch)
origin  https://github.com/Cal-Poly-RAMP/CARPOpenLane (push)
```

### Reproduction material

[issue_reproducible.partab.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11182043/issue_reproducible.partab.tar.gz)
[issue_reproducible.partaa.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11182044/issue_reproducible.partaa.tar.gz)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: runs/RUN_2023.04.07_23.01.57/logs/routing/15-resizer_design.log
[ERROR]: Last 10 lines:

[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2147483647
[INFO GRT-0002] Maximum degree: 1
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_design.tcl, 45 RSZ-0005
child process exited abnormally

12-cts.log
[INFO CTS-0007] Net "CLK" found for clock "CLK".
[WARNING CTS-0041] Net "CLK" has 0 sinks. Skipping...
[WARNING CTS-0083] No clock nets have been found.
[INFO CTS-0008] TritonCTS found 0 clock nets.
[WARNING CTS-0082] No valid clock nets in the design.
[INFO]: Repairing long wires on clock nets...
[INFO RSZ-0058] Using max wire length 3048um.

15-resizer_design.log
[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2147483647
[INFO GRT-0002] Maximum degree: 1
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_design.tcl, 45 RSZ-0005
```

vijayank88, 2023-04-08:
Are you running automated script or running any manual step?

vijayank88, 2023-04-08:
@jackpanderson 
your design dont have any instance/std cell. Nothing is placed in layout other than filler/decap cells.
Nothing is there for routing. 
```
[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1                  0             0            0.00%             0 /  0 /  0
met1           2126036             0            0.00%             0 /  0 /  0
met2           2152622             0            0.00%             0 /  0 /  0
met3           1502153             0            0.00%             0 /  0 /  0
met4            828281             0            0.00%             0 /  0 /  0
met5            207624             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total          6816716             0            0.00%             0 /  0 /  0

```
Can you share your source verilog passed to the flow?

jackpanderson, 2023-04-09:
[src.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11186031/src.zip)

donn, 2023-04-10:
Mind also sharing the config file?

vijayank88, 2023-04-11:
@jackpanderson 
You're instantiating `OTTER_MCU` inside `OTTER_WRAPPER`. 
Flow work on flatten design only. 
So you can skip resizer step.
For wrapper design, refer this example : https://github.com/efabless/caravel_user_project/blob/main/openlane/user_project_wrapper/config.json

kareefardi, 2024-07-17:
Please open a new issue if you have any other questions.

closed 2024-07-17 by kareefardi

kishan5637, 2024-08-07:
> Please open a new issue if you have any other questions.

I am also facing the same problem in my design.  I am hardening design using OpenLane.
Here is the error message: 
[STEP 18]
[INFO]: Running Global Routing Resizer Design Optimizations (log: designs/Single_Cycle_Top/runs/RUN_2024.08.07_09.50.53/logs/routing/18-resizer_design.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: designs/Single_Cycle_Top/runs/RUN_2024.08.07_09.50.53/logs/routing/18-resizer_design.log
[ERROR]: Last 10 lines:

[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2147483647
[INFO GRT-0002] Maximum degree: 1
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_design.tcl, 52 RSZ-0005
child process exited abnormally.

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1                  0             0            0.00%             0 /  0 /  0
met1           1116707             0            0.00%             0 /  0 /  0
met2           1123985             0            0.00%             0 /  0 /  0
met3            787477             0            0.00%             0 /  0 /  0
met4            429247             0            0.00%             0 /  0 /  0
met5            113013             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total          3570429             0            0.00%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2147483647
[INFO GRT-0002] Maximum degree: 1
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_design.tcl, 52 RSZ-0005.

Verilog and SDC file:
[src.zip](https://github.com/user-attachments/files/16521811/src.zip)

config.json file:
[config.json](https://github.com/user-attachments/files/16521754/config.json)