The-OpenROAD-Project/OpenLane #1726: Generate OpenROAD gui script as part of results

xobs, 2023-04-10:
### Description

OpenROAD has a very powerful GUI that requires some setup in order to get it working. Notably, the GUI must be passed a series of files to take maximum advantage of the information produced during build synthesis.

OpenLane could generate a script file as part of the release process that can be passed directly to openroad.

### Proposal

As an example, I have a script that loads the db, liberty, sdc, and spef files from a project:

```
read_db results/routing/minimax.odb
read_liberty /si/conda/share/pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
read_sdc results/final/sdc/minimax.sdc
read_spef results/final/spef/minimax.spef
```

Such a script could be generated very easily as part of the final generation process.

maliberty, 2023-04-10:
Note that flow.tcl has a -gui option already but it needs some work to correctly setup timing.

kareefardi, 2023-04-10:
Does https://github.com/The-OpenROAD-Project/OpenLane/pull/1729 addresses your concerns?
@xobs you can run `flow.tcl -design spm -tag tag1 -gui` to open up a (partially) finished run for spm with tag `tag1`. The above PR also load spef and sdc.

@maliberty `estimate_parastics` needs a flag. I am not aware of a direct way to find out if global routing has been run so I chose to print a message.

closed 2023-04-10 by donn

xobs, 2023-04-10:
That does address it, mostly. I didn't know that `flow.tcl` can be used to inspect the results of a run, in addition to running a flow.

I think the bigger issue is that the `-gui` flag doesn't appear to be documented, which may be why I didn't know it existed: https://openlane.readthedocs.io/en/latest/search.html?q=-gui&check_keywords=yes&area=default

It might be a good idea to document the existance of that option, so people know to look for it.

maliberty, 2023-04-10:
@kareefardi you could either look at what step in the flow is being loaded and decide based on that or you could look in odb and see if there are any guides present.