The-OpenROAD-Project/OpenLane #1732: SYNTH_CHECKS_ALLOW_TRISTATE is not working.

hakan-demirli, 2023-04-10:
### Description

A gate level design using `sky130_fd_sc_hd__ebufn_2` and these environment variables fails at  synthesis stage [STEP 1]  due to multiple constant drivers error.
```json
    "SYNTH_READ_BLACKBOX_LIB": 1,
    "SYNTH_CHECKS_ALLOW_TRISTATE": 1
```

### Expected Behavior

It should pass. Tag `2023.03.12` is working without any problems

### Environment report

```shell
Kernel: Linux v5.15.0-69-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v23.0.3 (OK)
OpenLane Git Version: b43df386c586c67355af2b336b7501b437b46a1a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b43df38 2023-04-10T17:33:21+02:00 Fix GDS write fail by KLayout (#1731) - Harald Pretl -  (grafted, HEAD -> master, origin/master, origin/HEAD)
---
Git Remotes

origin	ssh://git@github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	ssh://git@github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[bus.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11192105/bus.zip)

### Relevant log output

```shell
Encountered check error:
Warning: multiple conflicting drivers for BUS.\TRI_BUS_A.Do0 [1]:
    port Z[0] of cell TRI_BUS_A.BUSO[1].OBUF0 (sky130_fd_sc_hd__ebufn_2)
    port Z[0] of cell TRI_BUS_B.BUSO[1].OBUF0 (sky130_fd_sc_hd__ebufn_2)
```

hakan-demirli, 2023-04-11:
Adding ```SYNTH_ELABORATE_ONLY``` did not solve the problem. Also, even if it solves the problem the ability to create hybrid design RTL+Gate level is gone if enabled.

kareefardi, 2023-04-11:
At the moment, this is a limitation (due to the way we attempt to recognize tristate logic) that I am going to work on. As a workaround, I recommend turning off synth checks using `QUIT_ON_SYNTH_CHECKS`

closed 2023-06-13 by kareefardi