The-OpenROAD-Project/OpenLane #1737: Generated LEF views should have ANTENNAGATEAREA for the input pins

mo-hosni, 2023-04-10:
### Description

OpenRoad's antenna checks uses `ANTENNAGATEAREA` defined in the standard cells input pins in the LEF view to determine whether or not antenna violations exist. Also, `repair_antennas` function which is the main solution for antenna violations works the same way.  Both work on an abstract (not hierarchal) DEF and the LEF views of the macros or standard cells included in that DEF.

So, if a macro does not have `ANTENNAGATEAREA` defined on its input pins, there will exist antenna violations on the nets connected to these pins and those violations will not be detected or repaired. 

### Proposal

One solution is to make anntenna checks and repairs work on a hierarchal DEFs; so, macros will not be abstracted, and every net will have `ANTENNAGATEAREA` on the pin of the standard cell it is connected to.

Another, and much easier, solution is to add `ANTENNAGATEAREA` parameters to the abstracted LEF of the macros depending on the standard cells connected to these pins. 

I did a naive and easy fix to this by adding an `ANTENNAGATEAREA` on every input pin on every macro in the design using the smallest gate area in the SCL. So, basically I just inserted this line on every input pin on every macro:
`    ANTENNAGATEAREA 0.126000 ;`
This was done using the following [script](https://github.com/efabless/caravel/blob/caravel-redesign-2-OpenLane-update/scripts/insert_ant_areas.py) developed by  ChatGPT.

kareefardi, 2023-04-18:
Magic is capable of writing this property if `MAGIC_WRITE_FULL_LEF` is set to 1. We should verify the values generated by magic. Then, we can always write full lef and merge these values to the abstract lef, until we can get magic to write this property if abstract lef view

kareefardi, 2023-06-26:
Fixed by https://github.com/The-OpenROAD-Project/OpenLane/pull/1778 and https://github.com/The-OpenROAD-Project/OpenLane/pull/1850

closed 2023-06-26 by kareefardi