The-OpenROAD-Project/OpenLane #1738: sta.tcl script, trying to show all scripts in max and min reports

eusanchez, 2023-04-10:
### Discussed in https://github.com/The-OpenROAD-Project/OpenLane/discussions/1717

<div type='discussions-op-text'>

<sup>Originally posted by **eusanchez** April  4, 2023</sup>
Hello everyone, I want to be able to see every min and max report in the log directory in the sta.tcl file, but in the scripts it seems like there is a command -group_count that is not making it possible, i try different ways like not putting any number (only shows one report), removing -group_count (not working at all), and substitute with other commands like "-all_violators" or "violators_only" but it seems not working, can someone help me please. 

Here is a sneak peak of min reports in the script, this file just in case is in scripts directory, in openroad subdirectory. 

puts "min_report"
puts "\n==========================================================================="
puts "report_checks -path_delay min (Hold)"
puts "============================================================================"
report_checks -path_delay min -fields {slew cap input nets fanout} -format full_clock_expanded -group_count 5
puts "min_report_end"</div>

eusanchez, 2023-04-10:
Hello, I´m using it with a adder of 8 bits I found. In the scripts directory, in the openroad subdirectory there is the script that generates the file named sta.tcl, this file is in charge of listing every max and min report, the file is this one:

```tcl
# Copyright 2020-2022 Efabless Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#      http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
if { $::env(RUN_STANDALONE) == 1 } {
    source $::env(SCRIPTS_DIR)/openroad/common/io.tcl
    if { $::env(CURRENT_ODB) != "0" } {
        read
    } else {
        read_libs
        read_lef $::env(MERGED_LEF)
        read_netlist
    }

    if { $::env(STA_PRE_CTS) == 1 } {
        unset_propagated_clock [all_clocks]
    } else {
        set_propagated_clock [all_clocks]
    }
}

set_cmd_units -time ns -capacitance pF -current mA -voltage V -resistance kOhm -distance um

if { [info exists ::env(CURRENT_SPEF)] } {
    read_spef $::env(CURRENT_SPEF)
}

puts "min_report"
puts "\n==========================================================================="
puts "report_checks -path_delay min (Hold)"
puts "============================================================================"
report_checks -path_delay min -fields {slew cap input nets fanout} -format full_clock_expanded -group_count 5
puts "min_report_end"

puts "max_report"
puts "\n==========================================================================="
puts "report_checks -path_delay max (Setup)"
puts "============================================================================"
report_checks -path_delay max -fields {slew cap input nets fanout} -format full_clock_expanded -group_count 5
puts "max_report_end"

puts "check_report"
puts "\n==========================================================================="
puts "report_checks -unconstrained"
puts "============================================================================"
report_checks -unconstrained -fields {slew cap input nets fanout} -format full_clock_expanded

puts "\n==========================================================================="
puts "report_checks --slack_max -0.01"
puts "============================================================================"
report_checks -slack_max -0.01 -fields {slew cap input nets fanout} -format full_clock_expanded
puts "check_report_end"

puts "parastic_annotation_check"
puts "\n==========================================================================="
puts "report_parasitic_annotation -report_unannotated"
puts "============================================================================"
report_parasitic_annotation -report_unannotated
[... 714 lines cut ...]
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             5.67e-05   5.96e-06   1.10e-10   6.26e-05  71.2%
Combinational          1.27e-05   1.27e-05   1.96e-10   2.54e-05  28.8%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  6.94e-05   1.87e-05   3.06e-10   8.80e-05 100.0%
                          78.8%      21.2%       0.0%
power_report_end
area_report

===========================================================================
 report_design_area
============================================================================
Design area 773 u^2 100% utilization.
area_report_end
[WARNING] Did not save OpenROAD database!
Writing SDF to /openlane/designs/adder_8b_pipeline/runs/only_synth/results/synthesis/adder_8b_pipeline.sdf...

``` 

As you can see you can only see 5 max reports and 5 min reports, I want to show every max and min report it possible exists, but I dont know what to replace -group_count with, I already try some things I mention earlier but it didn´t work, and I know there are more because if you change 5 to 7 it shows now 7 reports, I want them all, can you help me please.

kareefardi, 2023-04-11:
I am not aware of a way to show _all_ paths. Would that really be useful ?
There is room for improvement here were we can parameterize `-group_count` and use `report_check_types -violators -min_delay|-max_delay` to show all setup and hold violations. That way you can see all violators.

maliberty, 2023-04-11:
The number of paths is exponential so it is rarely useful beyond a trivial design.

Vinayakamk, 2023-08-11:
Then try creating the all 4 path groups and pass sort by slack CMd.
U will the path from start and end point with timing properties of that perticular path

kareefardi, 2024-07-17:
Close this as stale.

closed 2024-07-17 by kareefardi