The-OpenROAD-Project/OpenLane #1739: OpenLane crashes with segfault on step 26 (multi-corner STA) [and further on creating issue reproducible]

hpretl, 2023-04-11:
### Description

```
[STEP 26]
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: spm/runs/RUN_2023.04.11_13.54.36/logs/signoff/26-rcx_mcsta.min.log)...
[ERROR]: during executing sta script /foss/tools/openlane/2023.04/scripts/openroad/sta/multi_corner.tcl
[ERROR]: Log: spm/runs/RUN_2023.04.11_13.54.36/logs/signoff/26-rcx_mcsta.min.log
[ERROR]: Last 10 lines:

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details.
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
define_corners Slowest Typical Fastest
read_liberty -corner Slowest /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ss_100C_1v60.lib
read_liberty -corner Typical /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
read_liberty -corner Fastest /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib
Warning: /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib line 619, table template ccsn_ovrf not found.
child killed: segmentation violation

[ERROR]: Creating issue reproducible...
```

### Expected Behavior

No crash.

### Environment report

```shell
/foss/tools/openlane/2023.04 > python3 ./env.py issue-survey
Kernel: Linux v5.15.49-linuxkit
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 49c28da7bbfcf8596f0e2f98fffb7a70a8d6ed92
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

49c28da7 2023-04-11T01:24:54+02:00 `or_issue.py` doc update and fix (#1733) - Kareem Farid -  (HEAD, tag: 2023.04.11, origin/master, origin/HEAD, master)
97fc540e 2023-04-11T01:24:14+02:00 `pdk_structure.md` updates (#1734) - Kareem Farid -  ()
b43df386 2023-04-10T17:33:21+02:00 Fix GDS write fail by KLayout (#1731) - Harald Pretl -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

Not produced, as another crash happens here:

```
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

[ERR] The sta script /foss/tools/openlane/2023.04/scripts/openroad/sta/multi_corner.tcl does not appear to be in /foss/tools/openlane/2023.04/scripts/sta.
[ERROR]: Failed to package reproducible.
[ERROR]: Step(26:parasitics_sta) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {run_tcl_script -tool sta -no_consume /foss/tools/openlane/2023.04/scripts/openroad/sta/multi_corner.tcl -no_update_current -indexed_log /foss/designs/spm/runs/RUN_2023.04.11_13.54.36/logs/signoff/26-rcx_mcsta.min.log -save to=/foss/designs/spm/runs/RUN_2023.04.11_13.54.36/results/routing/mca/process_corner_min,noindex,sdf,lib} CALL {run_sta_script /foss/tools/openlane/2023.04/scripts/openroad/sta/multi_corner.tcl -no_update_current -indexed_log /foss/designs/spm/runs/RUN_2023.04.11_13.54.36/logs/signoff/26-rcx_mcsta.min.log -save to=/foss/designs/spm/runs/RUN_2023.04.11_13.54.36/results/routing/mca/process_corner_min,noindex,sdf,lib} CALL {run_sta -log /foss/designs/spm/runs/RUN_2023.04.11_13.54.36/logs/signoff/rcx_mcsta.min.log -process_corner min -multi_corner -save_to /foss/designs/spm/runs/RUN_2023.04.11_13.54.36/results/routing/mca/process_corner_min} CALL run_parasitics_sta CALL run_parasitics_sta_step CALL {run_non_interactive_mode -design spm}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 214)
    invoked from within
"run_tcl_script -tool sta -no_consume {*}$args"
    (procedure "run_sta_script" line 2)
    invoked from within
"run_sta_script $::env(SCRIPTS_DIR)/openroad/sta/multi_corner.tcl  -no_update_current {*}$arg_list"
    (procedure "run_sta" line 72)
    invoked from within
"run_sta -log $log_name -process_corner $process_corner -multi_corner  -save_to $directory"
    (procedure "run_parasitics_sta" line 41)
    invoked from within
"run_parasitics_sta"
    (procedure "run_parasitics_sta_step" line 9)
    invoked from within
"run_parasitics_sta_step"} -errorline 1
```

### Relevant log output

```shell
See above.
```

vijayank88, 2023-04-11:
@hpretl 
Can you try with manual steps to create test case based on this: https://openlane.readthedocs.io/en/latest/for_developers/using_or_issue.html#manually

hpretl, 2023-04-11:
I used this:

`python3 ../tools/openlane/2023.04/scripts/or_issue.py --tool openroad --script ../tools/openlane/2023.04/scripts/openroad/sta/multi_corner.tcl spm/runs/foo/results/routing/spm.def`

to produce this artifact: [foo_multi_corner_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11200277/foo_multi_corner_packaged.tar.gz)

vijayank88, 2023-04-11:
@kareefardi 
Can you please check this
```
read_liberty -corner Fastest /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib
Warning: /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib line 619, table template ccsn_ovrf not found.
child killed: segmentation violation
```

hpretl, 2023-04-11:
Without the `-corner` I get this segfault:

```
openroad> read_liberty /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib
[WARNING STA-0122] /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib line 619, table template ccsn_ovrf not found.
Signal 11 received
Stack trace:
 0# 0x0000AAAAAC7DD5B4 in openroad
 1# __kernel_rt_sigreturn in linux-vdso.so.1
 2# sta::OutputCurrentWaveform::checkAxes(sta::TableTemplate*) in openroad
 3# sta::LibertyReader::beginVector(sta::LibertyGroup*) in openroad
 4# sta::libertyGroupBegin(char const*, sta::Vector<sta::LibertyAttrValue*>*, int) in openroad
 5# LibertyParse_parse() in openroad
 6# sta::parseLibertyFile(char const*, sta::LibertyGroupVisitor*, sta::Report*) in openroad
 7# sta::readLibertyFile(char const*, bool, sta::Network*) in openroad
 8# sta::Sta::readLibertyFile(char const*, sta::Corner*, sta::MinMaxAll const*, bool) in openroad
 9# sta::Sta::readLiberty(char const*, sta::Corner*, sta::MinMaxAll const*, bool) in openroad
10# 0x0000AAAAACA39EEC in openroad
11# TclNRRunCallbacks in /lib/aarch64-linux-gnu/libtcl8.6.so
12# 0x0000FFFFBA1FCF80 in /lib/aarch64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /lib/aarch64-linux-gnu/libtcl8.6.so
14# ord::tclAppInit(Tcl_Interp*) in openroad
15# Tcl_MainEx in /lib/aarch64-linux-gnu/libtcl8.6.so
16# main in openroad
17# 0x0000FFFFB66573FC in /lib/aarch64-linux-gnu/libc.so.6
18# __libc_start_main in /lib/aarch64-linux-gnu/libc.so.6
19# _start in openroad
Segmentation fault
```

kareefardi, 2023-04-11:
Do this happen without edit in spm configuration?

kareefardi, 2023-04-11:
I can confirm the issue with failing to package a reproducible. I will have a fix shortly then we can try out the reproducible.

hpretl, 2023-04-11:
BTW, in the same Docker container (same tool versions and all), OpenLane2 works like a charm.

hpretl, 2023-04-12:
I tested OL `2023.04.12`, same crash on multi-corner STA, but reproducible produced... here you go:

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11208606/issue_reproducible.tar.gz)

kareefardi, 2023-04-12:
Can you paste the header of the log file? 
```
OpenSTA 2.4.0 6840b7481d Copyright (c) 2021, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details.
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
```

hpretl, 2023-04-12:
```
OpenSTA 2.4.0 20805c83cf Copyright (c) 2021, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details.
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
```

hpretl, 2023-04-12:
Do you mean this (see above) or something else?

kareefardi, 2023-04-12:
Yes thanks I will take a look

kareefardi, 2023-04-12:
I still can't replicate this. Few (weird) things to note:
- This is OpenSTA version embedded in OpenLane `2023.04.12` is:
  ```
  > docker run -it efabless/openlane:2023.04.12
  OpenLane Container (23e367a):/openlane$ sta
  OpenSTA 2.4.0 6840b7481d Copyright (c) 2021, Parallax Software, Inc.
  License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>
  ```
- `6840b7481d` is a commit in OpenROAD repo itself. While `20805c83cf`, in the post above, belongs to OpenSTA repo. 
- I tried building OpenSTA `20805c83cf` separately and I still couldn't replicate the crash.

hpretl, 2023-04-12:
If you want to test in our container, I could push it for you. One note though: I am currently running everything on `aarch64`, maybe only this triggers the fail?

kareefardi, 2023-04-12:
Maybe. Yes please put the container

hpretl, 2023-04-12:
It is pushed, get it with `docker pull hpretl/iic-osic-tools:next`

kareefardi, 2023-04-12:
I think I understand now and the issue is rather simple and honestly don't know how I kept missing this for so long: The crash was fixed in original STA repo (https://github.com/The-OpenROAD-Project/OpenSTA/commit/974d9f43e7ba15a56f8534b36d7a107ef690602d) since it take a while for it to propagate to OpenROAD head it got ported to OpenLane here https://github.com/The-OpenROAD-Project/OpenLane/blob/23e367a926fd4a2e485e9edb908554a1ea2fa614/docker/openroad_app/Dockerfile#L78. The current version of OpenROAD in OpenLane is behind and doesn't have that patch - unless applied manually (I also confused this patch with another one that I have removed from OpenLane recently). Sorry for wasting a bit of time chasing this down.

hpretl, 2023-04-12:
Good find! I will try the patch.

hpretl, 2023-04-12:
Confirmed: The above patch solves the issue! Thanks for the help!

closed 2023-04-12 by hpretl

fazliemre, 2024-05-20:
Hello, ı have get error when i run the flow. How can i fix it?
I have added the Error page pic.

Error: multi_corner.tcl line 43, cannot open '/openlane/designs/i2c_master_top/src/i2c_master_bit_ctrl.v /openlane/designs/i2c_master_top/src/i2c_master_defines.v /openlane/designs/i2c_master_top/src/i2c_master_top.v /openlane/designs/i2c_master_top/src/timescale.v /openlane/designs/i2c_master_top/src/i2c_master_byte_ctrl.v'.

[image]

[image]