The-OpenROAD-Project/OpenLane #1740: OpenRoad order_wires fails in antenna_repair in global routing.

mo-hosni, 2023-04-11:
### Description

The following error occurs frequently in OpenRoad `antenna_repair` in global routing.

```
[ERROR ODB-0390] order_wires failed: net some_net, shorts to another term at wire point (2894780 762450)
Error: groute.tcl, 41 ODB-0390
```

### Expected Behavior

antenna_repair and global routing finish successfully 

### Environment report

```shell
Kernel: Linux v4.19.0-21-cloud-amd64
Distribution: debian 10
Python: v3.7.3 (OK)
Container Engine: docker v20.10.15 (OK)
OpenLane Git Version: d70884942e62945c564c7c5c23daea9d7ae58cce
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)                                                                                                                                      
                                                                                                                                                              
d7088494 2023-03-27T22:28:50+02:00 CI Updates (#1690) - Mohamed Gaber -  (HEAD, tag: 2023.03.28)                                                              
2e09573f 2023-03-27T20:30:59+02:00 Diode Insertion Overhaul (#1686) - Kareem Farid -  ()                                                                      
f4c6f5fe 2023-03-23T15:55:57+02:00 [BOT] Update openroad_app (#1681) - Openlane Bot -  (tag: 2023.03.24)                                                      
---                                                                                                                                                           
Git Remotes                                                                                                                                                   
                                                                                                                                                              
origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)                                                                                              
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

The file is larger than 25MB hence not allowed to upload. Here is a link:
https://drive.google.com/file/d/1IRLwy0k1ZYwdwSc9FyjLtwSEmUL2PuqY/view?usp=sharing

### Relevant log output

```shell
[INFO GRT-0006] Repairing antennas, iteration 3.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 57 antenna violations.
[INFO GRT-0015] Inserted 674 diodes.
[INFO GRT-0054] Using detailed placer to place 167 diodes.
[INFO GRT-0009] rerouting 468 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 74
[INFO GRT-0006] Repairing antennas, iteration 4.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[ERROR ODB-0390] order_wires failed: net mgmt_buffers.mprj2_vdd_logic1, shorts to another term at wire point (2894780 762450)
Error: groute.tcl, 41 ODB-0390
```

vijayank88, 2023-04-12:
In caravel wrapper, as I've seen most of resizer, repair_antenna checks all skipped.
Normally harden the macros and do top level connection in wrapper.
How you're implementing it ? @mo-hosni

vijayank88, 2023-04-12:
@mo-hosni 
Update in following variables, able to complete global routing stage without any error.
```
set ::env(GRT_ADJUSTMENT) {0.1};
set ::env(GRT_ALLOW_CONGESTION) {0};
```

mo-hosni, 2023-04-12:
> In caravel wrapper, as I've seen most of resizer, repair_antenna checks all skipped. Normally harden the macros and do top level connection in wrapper. How you're implementing it ? @mo-hosni

Do you mean caravel the top-level?
caravel just consists of caravel_core and the padframe (chip_io) placed without any routing or any kind of cell insertions. The IOs and power pins are connected by abutment and that's it.

mo-hosni, 2023-04-12:
> @mo-hosni Update in following variables, able to complete global routing stage without any error with latest OpenROAD HEAD.
> 
> ```
> set ::env(GRT_ADJUSTMENT) {0.1};
> set ::env(GRT_ALLOW_CONGESTION) {0};
> ```

I was also able to complete global routing by only changing `::env(GRT_ANT_MARGIN)` from `15` to `12`.
But, this failure happens frequently and always requires some calibration in the configuration, isn't that an issue?

vijayank88, 2023-04-12:
@eder-matheus 
Can you please check this antenna issue?

eder-matheus, 2023-04-19:
@mo-hosni I have these errors when opening your attached files:
```
tar -xf issue_reproducible.partaa.tar.gz
issue_reproducible/tmp/routing/27-diodes_legalized.obs.odb: truncated gzip input
tar: Error exit delayed from previous errors.
```
```
tar -xf issue_reproducible.partab.tar.gz
tar: Error opening archive: Unrecognized archive format
```

Could you attach a new file?

maliberty, 2023-04-19:
Does tar -zxf work?

eder-matheus, 2023-04-19:
> Does tar -zxf work?

Not really, same error messages from before.

vijayank88, 2023-04-20:
Merge into single file and extract again
 It works 
```
cat issue* > issue.tar.gz
tar -xzvf issue.tar.gz
```

eder-matheus, 2023-04-21:
@mo-hosni I ran your test case locally, and the global routing completes with the latest master branch. I also saw lots of warning messages about pins outside the die area. Perhaps it is related?

Could you share, If you have, another case with the issue?

eder-matheus, 2023-04-24:
> @mo-hosni I ran your test case locally, and the global routing completes with the latest master branch. I also saw lots of warning messages about pins outside the die area. Perhaps it is related?
> 
> Could you share, If you have, another case with the issue?

Nevermind, I was able to reproduce it running on a centos machine. For some reason, I can't replicate it on MacOS.

closed 2023-04-24 by maliberty

mo-hosni, 2023-04-26:
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11327708/issue_reproducible.tar.gz)

Just had the issue again. here is another reproducible of a completely different design and the design is very simple.

mo-hosni, 2023-04-26:
You fixed it!

Okay, thanks. I will check it out.

eder-matheus, 2023-04-26:
@mo-hosni Yes, we've merged a fix for the previous testcase you provided. But let me know if you find any other issues, order_wires is a trick function and can lead to more erros.