The-OpenROAD-Project/OpenLane #1742: Why is OpenSTA not managed via `tool_metadata.yml`?

hpretl, 2023-04-11:
### Description

Several tool dependencies of OpenLane are version-managed via [tool_metadata.yml](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/dependencies/tool_metadata.yml), but not OpenSTA.

Is there a specific reason for that?

### Expected Behavior

All relevant tools (potentially thus also OpenSTA) should be part of the build manifest.

### Environment report

```shell
n/a
```

### Reproduction material

n/a

### Relevant log output

```shell
n/a
```

hpretl, 2023-04-11:
OK, I am bit confused. We are building OpenSTA stand-alone in our Docker container, but there is also OpenSTA which is built as part of OpenROAD. It seems that only one OpenSTA instance should be there, so I guess the one as part of OpenROAD should be used. 

Can someone confirm this?

kareefardi, 2023-04-11:
Yes OpenSTA is part of OpenROAD build and this is what we are using in OpenLane

hpretl, 2023-04-11:
Thanks, we will remove the stand-alone OpenSTA build then.

closed 2023-04-11 by hpretl

donn, 2023-04-12:
Yeah, it is as you said. OpenROAD produces an `sta` binary already, so there was no need to manage yet another dependency.