The-OpenROAD-Project/OpenLane #1744: Error in Global routing stage

18341A04A6, 2023-04-12:
### Description

In global routing stage, net not properly covered error is being shown.
[image]

### Expected Behavior

Global routing should pass without errors

### Environment report

```shell
uploaded above
```

### Reproduction material

/content/runs/RUN_2023.04.12_09.27.55/issue_reproducible

### Relevant log output

```shell
uploaded above
```

vijayank88, 2023-04-12:
If the guidelines not followed it will closed without any response. Read carefully and update each column.

Which OpenLane commit you're working? it is missing in Environment setup.

`Reproduction material`, you should upload the `issue_reproducible` as a compressed file in `.zip` or `.tar.gz`

18341A04A6, 2023-04-12:
unable to download issue_reproducible . I am running openlane flow in google colab

vijayank88, 2023-04-12:
Follow this instruction: https://predictivehacks.com/?all-tips=how-to-download-files-and-folders-from-colab

18341A04A6, 2023-04-12:
I updated and sent my files in zip format.

vijayank88, 2023-04-12:
your zip file only has
```
DWT.ipynb
```

kareefardi, 2024-07-17:
Closing this as stale. Please file a new issue if needed.

closed 2024-07-17 by kareefardi