The-OpenROAD-Project/OpenLane #1746: [ERROR DPL-0036] Detailed placement failed.

tanvi-shewale, 2023-04-12:
### Description

I am getting the following error at the CTS phase.

[INFO]: Running Single-Corner Static Timing Analysis (log: designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/logs/placement/11-sta.log)...
[STEP 12]
[INFO]: Running Clock Tree Synthesis (log: designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/logs/cts/12-cts.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/cts.tcl
[ERROR]: Log: designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/logs/cts/12-cts.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  output46
[INFO DPL-0035]  _289_
[INFO DPL-0035]  _289_
[INFO DPL-0035]  _289_
[INFO DPL-0035]  output36
[INFO DPL-0035]  output36
[INFO DPL-0035]  output36
[ERROR DPL-0036] Detailed placement failed.
Error: cts.tcl, 67 DPL-0036
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

### Expected Behavior

Flow should proceed.

### Environment report

```shell
OpenLane Container (b43df38):/openlane$ python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.90.1-microsoft-standard-WSL2
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: b43df386c586c67355af2b336b7501b437b46a1a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b43df38 2023-04-10T17:33:21+02:00 Fix GDS write fail by KLayout (#1731) - Harald Pretl -  (grafted, HEAD -> master, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11215444/issue.tar.gz)

### Relevant log output

```shell
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/logs/placement/11-sta.log)...
[STEP 12]
[INFO]: Running Clock Tree Synthesis (log: designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/logs/cts/12-cts.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/cts.tcl
[ERROR]: Log: designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/logs/cts/12-cts.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  output46
[INFO DPL-0035]  _289_
[INFO DPL-0035]  _289_
[INFO DPL-0035]  _289_
[INFO DPL-0035]  output36
[INFO DPL-0035]  output36
[INFO DPL-0035]  output36
[ERROR DPL-0036] Detailed placement failed.
Error: cts.tcl, 67 DPL-0036
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

[ERROR]: Step(12:cts) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/cts.tcl -indexed_log $log -save "to=$::env(cts_results),noindex,def,sdc,odb""
    (procedure "run_cts" line 21)
    invoked from within
"run_cts"
    (procedure "run_cts_step" line 8)
    invoked from within
"run_cts_step"} -errorline 1
[INFO]: Saving current set of views in 'designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/decoder_logic/runs/RUN_2023.04.12_19.58.20/reports/metrics.csv'.
[INFO]: Saving runtime environment...
```

maliberty, 2023-04-12:
How many buffers were inserted during timing repair, eg:

Repair setup and hold violations...
[INFO RSZ-0040] Inserted 7 buffers.

Its possible that too much fixing is making the design too dense.

tanvi-shewale, 2023-04-12:
Where will I find this data?

maliberty, 2023-04-12:
In one of the log files.  I haven't run OL in a while so I'm not sure which.

tanvi-shewale, 2023-04-12:
I am not able to find these. Maybe the buffers are inserted at a later stage?

antonblanchard, 2023-04-12:
After adding the clock tree there is just not enough room. Likely you need to increase your die size.

[image]

maliberty, 2023-04-12:
Perhaps.  Can you open the results from the step before in the GUI and look at the placement density?

tanvi-shewale, 2023-04-12:
The die size is not a tight constraint right now. How do I increase the area?

tanvi-shewale, 2023-04-12:
I was able to increase the area and now the CTS phase is running fine. But it is getting stuck at step 40 where I get setup violations. I tried decreasing the clock frequency but the slack did not change. Any reason why this is happening or a possible way to fix it?

vijayank88, 2023-04-13:
@tanvi-shewale Share your `config.json` and I'll help you on how to increase die size

vijayank88, 2023-04-13:
Incremental variable will be overwritten with previous value.

Line 3: "CLOCK_PERIOD": 20,
Overwritten with Line 
```
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10
        },
```
So try this way:
```
{
    "DESIGN_NAME": "decoder_logic",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "ref::$CLOCK_PORT",
    "FP_PDN_VOFFSET": 7,
    "FP_PDN_HOFFSET": 7,
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg",
    "FP_PDN_SKIPTRIM": true,
    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 1900 1600",
    "pdk::sky130*": {
        "FP_CORE_UTIL": 45,
        "CELL_PAD": 2,
        "PL_BASIC_PLACEMENT": 0,
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 20
        },
        "scl::sky130_fd_sc_hdll": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 8
        },
        "scl::sky130_fd_sc_ls": {
            "CLOCK_PERIOD": 10,
            "SYNTH_MAX_FANOUT": 5
        },
```

tanvi-shewale, 2023-04-13:
So I have to increment my clock period at each instance?

tanvi-shewale, 2023-04-13:
I am getting this waring 
[image]

Is this normal?

vijayank88, 2023-04-13:
Yes.

tanvi-shewale, 2023-04-13:
Is there a config setting to convert this into a macro so that I can use this multiple times as a part of other design?

vijayank88, 2023-04-13:
Check caravel_user_project from efabless

vijayank88, 2023-04-15:
@tanvi-shewale If issue resolved, plz close the issue.

tanvi-shewale, 2023-04-15:
I will check if it is working and update soon

tanvi-shewale, 2023-04-16:
Where can I find the spice files associated with the design? I wanted to integrate the design with an analog module and wanted the spice files

kareefardi, 2023-04-18:
You can find spice view in `<run_dir>/results/final/spi/lvs/<design_name>.spice

tanvi-shewale, 2023-04-18:
I want to duplicate my design as a macro and integrate it with analog modules. So I don't know how to do this

kareefardi, 2024-07-17:
@tanvi-shewale the design should be a macro and you should be able to use the GDS and SPICE models to integrate with analog modules. The opensource silicon slack workspace has channels very useful for analog development: https://open-source-silicon.dev/. Please ask there and open an new issue if needed.

closed 2024-07-17 by kareefardi