The-OpenROAD-Project/OpenLane #1751: Multi-thread to run the OpenROAD process maybe?

JiaDYuan, 2023-04-14:
### Description

Hi, I have a stupid question. I am running a relatively complicated project with approximate 800,000 gates. But it's running very slow.  Now it's still on run in the Global Routing stage after almost 2 days' running. And after checking the CPU resources utilization, I found out there is only one CPU core running the "openroad" command. So openroad is a single thread process? How can I make it run in multi CPU cores parallelly? Or it there any method to make the flow running faster? 

### Proposal

Improve the running speed to complete the a bit complex projects

hakan-demirli, 2023-04-14:
Some parts are parallelized like detailed routing and xor steps. 
```json
    "ROUTING_CORES": 8,
    "KLAYOUT_XOR_THREADS": 8,
```

donn, 2023-05-10:
Not really much we can do here. OpenROAD is indeed single-threaded for most things (notably excluding detailed routing.)

closed 2023-05-10 by donn