The-OpenROAD-Project/OpenLane #1752: Head Maps Power Density 0.00

Baungarten-CINVESTAV, 2023-04-15:
### Description

The OpenROAD -GUI Head Map Power DEnsity view shows 0.00 fW when loading .def and .lef files or when the .odb file is loaded

### Expected Behavior

1) Hardening a module with the OpenLANE workflow
2) Loading .def and .lef files in OpenROAD GUI
3) Showing the power density of Head Map

[image]

### Environment report

```shell
N/A
```

### Reproduction material

[user_proj_example.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11240861/user_proj_example.zip)

### Relevant log output

```shell
N/A
```

vijayank88, 2023-04-16:
Can you check the log/report file, report_power it has?

maliberty, 2023-04-16:
I'm guessing from:
[image]

that you haven't loaded the liberty files.

Baungarten-CINVESTAV, 2023-04-16:
@vijayank88 and @maliberty thanks for your quick answer.

I'm running the OpenLANE flow with Caravel, and then i export the ".def"  and ".lef" file in to "OpenROAD-flow-scripts".

With your answers i noticed two thinks first, the power report will be 0 W if the module its composed only for sub modules. Second as Matt says probably i haven't loaded the liberty files, i just import to OpenROAD the ".def" and ".lef" file. 

I attached you two different files:

- Top module power report & OpenROAD Power Density view
[image]
[Power reports top module.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11243021/Power.reports.top.module.zip)

- Sub module power report & OpenROAD Power Density view
[image]

[Sub Module Power report.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11243033/Sub.Module.Power.report.zip)

Now my question is how can i import those files ( liberty files) to OpenROAD-flow-scripts?

vijayank88, 2023-04-17:
@Baungarten-CINVESTAV Seems you're using 2 months old OpenLane commit.
Update your OpenLane to latest and run the flow.
Power density information loaded with `.odb` file to view in gui. Nothing can be done with `.rpt` file.
Load `.odb` and check the results.

Baungarten-CINVESTAV, 2023-04-20:
[image]

I installed the newest version of OpenROAD & OpenLANE, then i ran the PPU OpenLANE example, and finally i exported opened the ".odb" with OpenROAD -GUI and i had the same result.

What could be the problem?

vijayank88, 2023-04-20:
@Baungarten-CINVESTAV 
Follow below steps to view power density in OpenROAD GUI
```
% openroad
> read_liberty <path to>/sky130B/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
>read_db <path to>/routing/PPU.odb
>gui::show
```
From `Heat Maps > Power Density`  menu you can view expected results.

Baungarten-CINVESTAV, 2023-04-20:
Thanks, it works for OpenLANE direct workflow.

[image]

Maybe this is not the right issues but do you know why when I load a ".odb" file from caravel it still shows up with 0 fW, could it be the version of OpenLANE that Caravel uses?

vijayank88, 2023-04-20:
Have you load the `.lib` file?

Baungarten-CINVESTAV, 2023-04-20:
Yes, I followed the same steps
```
openroad
OpenROAD v2.0-7480-g8a4065b09 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_liberty sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
1
openroad> read_db user_project_wrapper.odb
openroad> gui::show
```

[image]

Baungarten-CINVESTAV, 2023-04-20:
well, just now i tried to see the power map of a sub-module, and it works. so i think the problem is when we see the power map of the core module.

kareefardi, 2024-07-17:
@Baungarten-CINVESTAV Please open a new issue if needed. I tried viewing PPU using the above instructions it worked.

closed 2024-07-17 by kareefardi