The-OpenROAD-Project/OpenLane #1753: Remove `-from` and `-to`

donn, 2023-04-16:
### Description

`-from` and `-to` were hacked together by an intern. OpenLane 1 was not designed with repeatable steps in mind, unlike its successor.

### Proposal

`-from` and `-to` should be removed from the documentation and the CLI flags should throw an error.

closed 2023-04-16 by kareefardi

mguthaus, 2023-05-12:
Can you clarify why this was dropped and not fixed if there was an issue? This seems like it should be a supported option.

donn, 2023-05-15:
As the prompt says, OpenLane 1 was not written with `-from` and `-to` in mind. Flow steps are not designed to be re-entrant and a variable set in a later step may permanently alter the behavior of a previous step. The effort to track down and fix all of this would be effort better spent on the rewrite (which WAS written with `-from` and `-to` in mind), so we simply chose to remove it from this version.

PR #1757 (2023-04-16): Remove `-from` and `-to`
- Removes `-from` and `-to`

---

Resolves #1753