The-OpenROAD-Project/OpenLane #1763: RePlAce divergence detected on small design

mattvenn, 2023-04-18:
### Description

On this design https://github.com/gbsha/ttfir/commit/0be0ce8776e6f3de70e78c45a77ea3d5f713f808
We are getting `RePlAce divergence detected. Re-run with a smaller max_phi_cof value.`

Yosys report is ~200 cells and ~2000 square microns area. We have a fixed size die of 25,000 square microns, so plenty of room.

### Expected Behavior

Flow should succceed

### Environment report

```shell
Kernel: Linux v5.15.0-69-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11261353/issue_reproducible.tar.gz)

### Relevant log output

```shell
[STEP 7]                                                                                       
[INFO]: Running Global Placement (log: ../work/runs/wokwi/logs/placement/7-global.log)...                                                                                                     
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl                   
[ERROR]: Log: ../work/runs/wokwi/logs/placement/7-global.log                                                                                                                                  
[ERROR]: Last 10 lines:                                                                                                                                                                       
[NesterovSolve] Iter: 630 overflow: 0.262545 HPWL: 5322542                                     
[NesterovSolve] Iter: 640 overflow: 0.249751 HPWL: 5560596                                     
[NesterovSolve] Iter: 650 overflow: 0.202928 HPWL: 5792007                                     
[NesterovSolve] Iter: 660 overflow: 0.176583 HPWL: 6933765                                     
[NesterovSolve] Iter: 670 overflow: 0.224019 HPWL: 6980485                                                                                                                                    
[NesterovSolve] Iter: 680 overflow: 0.192639 HPWL: 8696204                                     
[NesterovSolve] Iter: 690 overflow: 0.164443 HPWL: 11194311                                                                                                                                   
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.                                                                                                        
Error: gpl.tcl, 69 GPL-0307                                                                    
child process exited abnormally                                                                                                                                                               
                                                                                               
[ERROR]: Creating issue reproducible...                                                        
[INFO]: Saving runtime environment...                                                          
OpenLane TCL Issue Packager                                                                                                                                                                   
                                                                                                                                                                                              
EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD                                                                                                                
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING                                                                                                              
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR                                                                                                               
CONDITIONS OF ANY KIND.                                                                                                                                                                       
                                                                                                                                                                                              
BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER                                                                                                               
AND ALL IT ENTAILS.                                                                                                                                                                           
                                                                                                                                                                                              
Parsing config file(s)…                                                                                                                                                                       
Setting up /work/runs/wokwi/issue_reproducible…                                                                                                                                               
Done.
```

vijayank88, 2023-04-18:
@mattvenn 
This is resolved with latest OpenROAD HEAD. 
Can you try with latest OpenLane or OpenROAD commit?

mattvenn, 2023-04-18:
What was the issue?

mattvenn, 2023-04-18:
Is the issue resolved in a tagged version of OpenLane or only head?

vijayank88, 2023-04-18:
#1761 Updated the OpenROAD commit. So first try with updated OpenLane commit.
Same test case working fine in latest OpenROAD head as well.

mattvenn, 2023-04-18:
Any idea what caused the issue? That PR to change the OpenROAD commit doesn't help me much

mattvenn, 2023-04-18:
Unfortunately with the latest openlane tag the flow fails due to:

https://github.com/gbsha/ttfir/actions/runs/4732232801/jobs/8398140680#step:8:130

* Failed to create manufacturability and metric reports
* FileNotFoundError: [Errno 2] No such file or directory: '/work/src/runs/wokwi/config.tcl'
* Error: : /work/runs/wokwi/reports/signoff/31-rcx_sta.min.rpt doesn't exist.

So we have reverted to a previous version.

closed 2023-04-18 by mattvenn

kareefardi, 2023-04-18:
@mattvenn you should file a new issue.