The-OpenROAD-Project/OpenLane #1768: [ERROR]: Mismatch during LVS

akhilesh911, 2023-04-20:
### Description

Hello everyone,
I am new to OpenLANE and trying to rebuild a 16 bit processor using caravel user project. The repo is given [here](https://github.com/7vik-g/16bit-pipelined-RISC-processor-gf180.git). I am using gf180 PDK for this purpose. The design uses two macros, interface and processor, and four SRAM cells of gf180 PDK. There are mismatch in SRAM power pins, vdd and vss. But not such error in case of other macros. 

Net: processor/vdd                         |(no matching net)                          
  processor/vdd = 1                        |                                           
                                                       |                                           
Net: processor/vss                         |(no matching net)                          
  processor/vss = 1                        |                                           
---------------------------------------------------------------------------------------

---------------------------------------------------------------------------------------
Net: vss                                                                     |(no matching net)                          
  gf180mcu_fd_ip_sram__sram256x8m8wm1/VSS  |                                           
                                                                                  |                                           
Net: vdd                                                                     |(no matching net)                          
  gf180mcu_fd_ip_sram__sram256x8m8wm1/VDD  |                                           
The log files are attached. 
[23-lvs.lef.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11260264/23-lvs.lef.log)
[23-user_project_wrapper.lef.lvs.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11260266/23-user_project_wrapper.lef.lvs.log)

### Expected Behavior

No erros. Flow complete.

### Environment report

```shell
Kernel: Linux v5.4.0-146-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
```

### Reproduction material

No reproducible issues found.

### Relevant log output

```shell
[INFO]: Writing Verilog (log: ../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/logs/signoff/21-write_powered_verilog.log)...
[INFO]: Running LVS (log: ../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/logs/signoff/23-lvs.lef.log)...
[ERROR]: There are LVS errors in the design: See '../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/reports/signoff/23-user_project_wrapper.lvs.rpt' for a summary and '../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/logs/signoff/23-lvs.lef.log' for details.
[INFO]: Saving current set of views in '../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2023-04-20:
@akhilesh911 
Attach this log `/home/signoff/16bit-pipelined-RISC-processor-gf180/openlane/user_project_wrapper/runs/23_04_18_14_44/logs/signoff/23-lvs.lef.log`

akhilesh911, 2023-04-20:
I've already attached.

vijayank88, 2023-04-20:
@akhilesh911 
For gf180 check config.tcl: https://github.com/efabless/caravel_user_project/blob/gfmpw-0d/openlane/user_project_wrapper/config.tcl

`FP_PDN_MACRO_HOOKS` is
```
### Macro PDN Connections
set ::env(FP_PDN_MACRO_HOOKS) "\
	mprj vdd vss vdd vss"
```

vijayank88, 2023-04-20:
@d-m-bailey 
For GF180 power domains are `vdd vss` right?

akhilesh911, 2023-04-21:
@d-m-bailey I tried your suggestion. The pdn log gives warning that there is no via insertion between metal3 and metal4 for vdd and vss pins. Can you please suggest the solution?
[7-pdn.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11292232/7-pdn.log)

d-m-bailey, 2023-04-21:
@akhilesh911 probably not the best advice someone could give, but first try looking at the coordinates shown in the layout. Is there a metal3 power trunk in the macro that should connect to a metal4 power trunk in the top level?

Do you have metal4 obstruction layers over the macro?

Are your macros large enough or placed in the correct position that they intersect with the top power trunks?

akhilesh911, 2023-04-21:
@d-m-bailey The gf180 SRAM macro doesnot have metal4 obstruction layers, but the processor macro have for both metal3 and metal4. 
The coordinates for macros are
micro_processor 1000 1900 N
IO_interface 600 750 N
instr_memory_MSB 1000 100 N
instr_memory_LSB 1550 100 N
data_memory_MSB 100 100 N
data_memory_LSB 2450 100 N

d-m-bailey, 2023-04-21:
@akhilesh911 Sorry for the confusion. The pdn log has coordinates where vias cannot be placed. Can you open the gds and check those coordinates? You want to verify that metal3 and metal4 can be placed there. Does the SRAM macro get connected?

vijayank88, 2023-04-21:
@d-m-bailey 
For a note, GF180 has power pins as `VDD VSS`. Is that to be updated in `FP_PDN_MACRO_HOOKS`?

d-m-bailey, 2023-04-21:
@vijayank88 You may be right. It might be case sensitive.

akhilesh911, 2023-04-24:
@vijayank88 @d-m-bailey Sorry for late replying. The signals are mapped correctly in verilog file. so there is no error for case sensitive.

vijayank88, 2023-04-24:
@akhilesh911 Did you added `GRT_OBS` for SRAM macros?

akhilesh911, 2023-04-24:
> @akhilesh911 Did you added `GRT_OBS` for SRAM macros?

@vijayank88 no, i haven't added that.

d-m-bailey, 2023-04-24:
@akhilesh911 There should be a def file (and maybe a lef file) that you can open in the openroad gui, magic or maybe klayout. You should be able to check the error coordinates there.

akhilesh911, 2023-04-25:
@d-m-bailey For me, openroad read_def shows no technology file has been read, even i have read_lib file. so i am unable to view def. And klayout is throwing error 
[image]

kareefardi, 2023-04-26:
@akhilesh911 @vijayank88 is correct. 

> @d-m-bailey For a note, GF180 sram macros has power pins as `VDD VSS`. Is that to be updated in `FP_PDN_MACRO_HOOKS`?

`FP_PDN_MACRO_HOOKS` is a mapping between the design's power net(s) and a macro pin. Sram macro power pins are called `VDD` and `VSS`

kareefardi, 2023-04-26:
In your case, the pin name of the sram macros match the standard cells pin names so "macro hooks" will be automatically done for the sram whether what you define is correct or not. What is actually missing is:
```
add_pdn_connect \
    -grid macro \
    -layers "Metal3 Metal4"
```
You need to create a copy of OpenLane's `scripts/openroad/common.tcl` and point to it using `PDN_CFG` - which will be renamed to match the documentation -. Then add the above code block to your newly created file. Currently OpenLane assumes fixed power layers for macros

akhilesh911, 2023-04-27:
@kareefardi yes I understood. Along with this i have reconfigured some parameters, and the flow is passed. Thank you so much.

closed 2023-04-27 by akhilesh911

vyom-elan, 2024-01-17:
I am implementing a sample timer project and during hardening the user_project_wrapper module I am getting a LVS mismatch error, I am not understanding the issue since hardening of user_project_example was successful. I am attaching the log file any input will be appreciated. Also I am getting several warning related to the PDK. Could this be a source for an error? 
[image]
Thank you

vyom-elan, 2024-01-17:
LVS Log file of wrapper module - 
https://github.com/vyom-elan/caravel_uer_project/blob/main/user_project_wrapper.lef.lvs.log

d-m-bailey, 2024-01-17:
@vyom-elan It looks there are inputs tied high or low on the top level of the design.

Since you are not synthesizing the top level, you'll get `conb` cells that aren't connected to power (no cell rows, no fill cells, no decap cells, no standard cell power rails).

There are a couple solutions.
1) Tie the inputs at the `user_proj_example` level.
2) Remove the `user_proj_example` level by flattening the design. This probably requires a little more work because you have to do a major overhaul of the `config.json` file. It's option 2 [here](https://caravel-harness.readthedocs.io/en/latest/caravel-with-openlane.html).

vyom-elan, 2024-01-23:
@d-m-bailey  "It looks there are inputs tied high or low on the top level of the design." - I have interpreted it and checked it for any input set to logical high. Is that correct? Also I went with approach 2 and I am getting the following error during synthesis of user_project_wrapper. What could be the possible issue?
[image]

kareefardi, 2024-01-23:
@vyom-elan The error is about yosys failing to map some cells. Check the netlist generated by Yosys. If you would like, you can open a new issue.