The-OpenROAD-Project/OpenLane #1770: Make file not working

eusanchez, 2023-04-21:
### Description

Hello, I'm trying to install OpenLane I followed exactly the instructions as they are in the github page but when I try to do **### make**, it doesnt seems to work, seems i'm missing pdk directories, but I did everything the instructions said, so I don't know where I am doing wrong

`carlos@user:~/OpenLane$ make

make[1]: Entering directory '/home/carlos/OpenLane'

02a096e7af3f152b436b5abf3a1de0e278a5b63f: Pulling from efabless/openlane

Digest: sha256:3402c8a26bc93b4487bb71bec997bd3ab407ebae4f517d9b29f2e85ed5c292bc

Status: Image is up to date for efabless/openlane:02a096e7af3f152b436b5abf3a1de0e278a5b63f

docker.io/efabless/openlane:02a096e7af3f152b436b5abf3a1de0e278a5b63f

make[1]: Leaving directory '/home/carlos/OpenLane'

PYTHONPATH= ./venv/bin/python3 -m pip install --upgrade --no-cache-dir volare

Requirement already satisfied: volare in ./venv/lib/python3.10/site-packages (0.7.1)

Requirement already satisfied: click<9,>=8.0.0 in ./venv/lib/python3.10/site-packages (from volare) (8.1.3)

Requirement already satisfied: pyyaml<7,>=5 in ./venv/lib/python3.10/site-packages (from volare) (5.4.1)

Requirement already satisfied: rich<13,>=12 in ./venv/lib/python3.10/site-packages (from volare) (12.6.0)

Requirement already satisfied: requests<3,>=2.27.0 in ./venv/lib/python3.10/site-packages (from volare) (2.28.2)

Requirement already satisfied: pcpp<2,>=1.2 in ./venv/lib/python3.10/site-packages (from volare) (1.30)

Requirement already satisfied: charset-normalizer<4,>=2 in ./venv/lib/python3.10/site-packages (from requests<3,>=2.27.0->volare) (3.1.0)

Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.10/site-packages (from requests<3,>=2.27.0->volare) (3.4)

Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./venv/lib/python3.10/site-packages (from requests<3,>=2.27.0->volare) (1.26.15)

Requirement already satisfied: certifi>=2017.4.17 in ./venv/lib/python3.10/site-packages (from requests<3,>=2.27.0->volare) (2022.12.7)

Requirement already satisfied: commonmark<0.10.0,>=0.9.0 in ./venv/lib/python3.10/site-packages (from rich<13,>=12->volare) (0.9.1)

Requirement already satisfied: pygments<3.0.0,>=2.6.0 in ./venv/lib/python3.10/site-packages (from rich<13,>=12->volare) (2.15.1)

./venv/bin/volare enable --pdk sky130

[22:02:47] Found version 0c37b7c76527929abfbdbd214df4bffcd260bf50 in ./dependencies/tool_metadata.yml.                                                                                        common.py:154

Downloading pre-built tarball for 0c37b7c76527929abfbdbd214df4bffcd260bf50… ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00

Unpacking…                                                                  ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--

Traceback (most recent call last):

  File "/usr/lib/python3.10/pathlib.py", line 1175, in mkdir

    self._accessor.mkdir(self, mode)

FileNotFoundError: [Errno 2] No such file or directory: '/home/carlos/.volare/volare/sky130/versions/0c37b7c76527929abfbdbd214df4bffcd260bf50/sky130B/libs.tech/magic'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "/usr/lib/python3.10/pathlib.py", line 1175, in mkdir

    self._accessor.mkdir(self, mode)

FileNotFoundError: [Errno 2] No such file or directory: '/home/carlos/.volare/volare/sky130/versions/0c37b7c76527929abfbdbd214df4bffcd260bf50/sky130B/libs.tech'

[... 468 lines cut ...]

    return pathlib.Path(path).mkdir(parents=True, exist_ok=True)

  File "/usr/lib/python3.10/pathlib.py", line 1179, in mkdir

    self.parent.mkdir(parents=True, exist_ok=True)

  File "/usr/lib/python3.10/pathlib.py", line 1179, in mkdir

    self.parent.mkdir(parents=True, exist_ok=True)

  File "/usr/lib/python3.10/pathlib.py", line 1179, in mkdir

    self.parent.mkdir(parents=True, exist_ok=True)

  [Previous line repeated 1 more time]

  File "/usr/lib/python3.10/pathlib.py", line 1175, in mkdir

    self._accessor.mkdir(self, mode)

PermissionError: [Errno 13] Permission denied: '/home/carlos/.volare/volare/sky130/versions'

make: *** [Makefile:127: pdk] Error 1
```

vijayank88, 2023-04-24:
@eusanchez 
Maybe it is related to  network issue. Try again following:
```
cd OpenLane
git checkout master
git pull
make
```

kareefardi, 2023-04-26:
Does this error happen consistently ?

ensariskin, 2023-06-21:
I have the same problem, is there any solution? Thanks.

ensariskin, 2023-06-21:
> I have the same problem, is there any solution? Thanks.

I have solved the problem by changing folder permission of .volare folder. After making this folder accessible by normal user, make command finished successfully.

closed 2023-06-23 by vijayank88