The-OpenROAD-Project/OpenLane #1771: Treating unmatched pins as errors

bgatusch, 2023-04-21:
### Description

Hi, I am new to OpenLane and have been following the MPW-6 Walkthrough tutorial using the same RTL files in the walkthrough example. However, I am encountering the warning below when attempting to harden the design, which is flagged as an error.
I notice in the tutorial there is a config.tcl that has been edited. However, this isn't present in my folder, so I have edited the config.json instead. Which file(s) have I missed editing?

**Warning: Some pins weren't matched by the config file
Those are: ['clk', 'reset', 'led_out[0]', 'led_out[1]', 'led_out[2]', 'led_out[3]', 'led_out[4]', 'led_out[5]', 'led_out[6]']
Treating unmatched pins as errors. Exiting..
child process exited abnormally**

MPW-6 Walkthrough tutorial:
https://www.youtube.com/watch?v=MNuoYz_MM-c&t=612s

### Expected Behavior

Be able to harden my design.

### Environment report

```shell
bgatusch@BG3W10L:~/caravel_test/dependencies/openlane_sr$ python3 env.py issue-survey
Kernel: Linux v5.15.90.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.23 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
bgatusch@BG3W10L:~/caravel_test/dependencies/openlane_sr$
```

### Reproduction material

bgatusch@BG3W10L:~/caravel_test$ time make seven_segment_seconds
make -C openlane seven_segment_seconds
make[1]: Entering directory '/home/bgatusch/caravel_test/openlane'
# seven_segment_seconds
mkdir -p ./seven_segment_seconds/runs/23_04_21_19_28
rm -rf ./seven_segment_seconds/runs/seven_segment_seconds
ln -s $(realpath ./seven_segment_seconds/runs/23_04_21_19_28) ./seven_segment_seconds/runs/seven_segment_seconds
docker run -it -v $(realpath /home/bgatusch/caravel_test/..):$(realpath /home/bgatusch/caravel_test/..) -v /home/bgatusch/caravel_test/dependencies/pdks:/home/bgatusch/caravel_test/dependencies/pdks -v /home/bgatusch/caravel_test/caravel:/home/bgatusch/caravel_test/caravel -v /home/bgatusch/caravel_test/dependencies/openlane_src:/openlane -v /home/bgatusch/caravel_test/mgmt_core_wrapper:/home/bgatusch/caravel_test/mgmt_core_wrapper -e PDK_ROOT=/home/bgatusch/caravel_test/dependencies/pdks -e PDK=sky130A -e MISMATCHES_OK=1 -e CARAVEL_ROOT=/home/bgatusch/caravel_test/caravel -e OPENLANE_RUN_TAG=23_04_21_19_28 -e MCW_ROOT=/home/bgatusch/caravel_test/mgmt_core_wrapper -u 1000:1000 \
        efabless/openlane:2023.02.23 sh -c "flow.tcl -design $(realpath ./seven_segment_seconds) -save_path $(realpath ..) -save -tag 23_04_21_19_28 -overwrite -ignore_mismatches"
OpenLane a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in '../home/bgatusch/caravel_test/openlane/seven_segment_seconds/config.json'...
[INFO]: PDK Root: /home/bgatusch/caravel_test/dependencies/pdks
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /home/bgatusch/caravel_test/openlane/seven_segment_seconds/runs/23_04_21_19_28
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[STEP 1]
[INFO]: Running Synthesis (log: ../home/bgatusch/caravel_test/openlane/seven_segment_seconds/runs/23_04_21_19_28/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: ../home/bgatusch/caravel_test/openlane/seven_segment_seconds/runs/23_04_21_19_28/logs/synthesis/2-sta.log)...
[... 66 lines cut ...]
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
OpenROAD 1f720d3b442e2cd8dc6c5372535320b18a105e8d
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Top-level design name: seven_segment_seconds
Warning: Some pins weren't matched by the config file
Those are: ['clk', 'reset', 'led_out[0]', 'led_out[1]', 'led_out[2]', 'led_out[3]', 'led_out[4]', 'led_out[5]', 'led_out[6]']
Treating unmatched pins as errors. Exiting..
child process exited abnormally

[INFO]: Saving current set of views in '../home/bgatusch/caravel_test/openlane/seven_segment_seconds/runs/23_04_21_19_28/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../home/bgatusch/caravel_test/openlane/seven_segment_seconds/runs/23_04_21_19_28/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../home/bgatusch/caravel_test/openlane/seven_segment_seconds/runs/23_04_21_19_28/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
make[1]: *** [Makefile:73: seven_segment_seconds] Error 255
make[1]: Leaving directory '/home/bgatusch/caravel_test/openlane'
make: *** [Makefile:107: seven_segment_seconds] Error 2

real    0m9.041s
user    0m0.143s
sys     0m0.033s
```

d-m-bailey, 2023-04-22:
@bgatusch Looks like you need to change the `openlane/seven_segment_seconds/pin_config.tcl` to only the ports in `verilog/rtl/seven_segment_seconds.v`. In the video, this file is only 49 bytes. 

@mattvenn with `config.tcl`, was this file created automatically or is there a step missing somewhere.

mattvenn, 2023-04-24:
I'm pretty sure I didn't do anything special.
Here's the file in the repo: https://github.com/mattvenn/caravel_walkthrough/blob/walkthrough/openlane/seven_segment_seconds/pin_order.cfg

d-m-bailey, 2023-04-24:
@bgatusch Can you check that you're using the `pin_order.cfg` file shown above?

kareefardi, 2023-04-26:
As the error says, the mentioned pins aren't in the pins config file. The default behavior ,at the time of the video, was to randomly place pins that aren't specified. This changed in https://github.com/The-OpenROAD-Project/OpenLane/pull/1086 and can be controlled by the variable `FP_IO_UNMATCHED_ERROR`

mattvenn, 2023-04-26:
Ahh, thanks. That will be it then.

vijayank88, 2023-04-26:
User has to use default random pin placement or if they're using `pin_order.cfg` it is mandatory to provide all pin list in the file. If missing something the `FP_IO_UNMATCHED_ERROR` will fail with error.

@bgatusch add following list in `pin_order.cfg` and continue
```
['clk', 'reset', 'led_out[0]', 'led_out[1]', 'led_out[2]', 'led_out[3]', 'led_out[4]', 'led_out[5]', 'led_out[6]']
```

bgatusch, 2023-04-26:
Hello, thank you very much for your reply. Editing the pin_order.cfg to match the ports in my design, as shown above corrected this error and I am able to harden the design. 

I do have a couple of other questions, do I need to open new issues for them individually?
I am writing them here (although can also create a new issue). My questions are:

1. After I harden the user wrapper, this completes the flow successfully. However, I do get the following warning, which I have not been able to clear by changing the CLOCK_PERIOD in the config.json file. I also tried repositioning the design within the wrapper and changing the die area. 
"[WARNING]: There are max slew violations in the design at the typical corner. Please refer to '../home/bgatusch/caravel_test/openlane/user_project_wrapper/runs/23_04_26_11_38/reports/signoff/21-rcx_sta.slew.rpt'."

2. What is the highest clock frequency (smallest clock period) the design could be run at? 
From the config.json, changing the CLOCK_PERIOD within the first few lines of the file doesn't seem to affect the CLOCK_PERIOD reported in "metrics.csv" under the runs -> reports folder. However, changing it from this section does. However, are we allowed to modify it there? (Suggested max. frequency for the library below seems to be 100 MHz.)
    "pdk::sky130*": {
        "FP_CORE_UTIL": 45,
        "RT_MAX_LAYER": "met4",
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10
        },

3. In the metrics.csv file all power information seem to report "-1". For examples:
power_slowest_internal_uW : -1
power_slowest_switching_uW :-1
power_slowest_leakage_uW : -1
power_typical_internal_uW : -1
..etc.
However, there is some power information in files like 7-gpl_sta.parasitics_check.rpt under openlane -> runs -> date -> reports -> placement. I wasn't sure why this info isn't reflected in the metrics.csv file.

Please let me know if I need to report each of these questions separately. 
My original issue has been solve, thank you.

vijayank88, 2023-04-29:
Solution:
1. Ignore warnings, until error or failure occurs
2. Until its meeting timing, you can tighten clock period.
3. `-1` means its empty or not reported.

closed 2023-04-29 by vijayank88