The-OpenROAD-Project/OpenLane #1772: ./flow -gui crashing with segfault

mattvenn, 2023-04-26:
### Description

After hardening a design with openlane:

`./flow.tcl -design spm`

I try to open the gui (which worked about a month ago):

`./flow.tcl -design spm -tag RUN_2023.04.26_09.09.28 -gui`

And get a segfault.

### Expected Behavior

The GUI should open

### Environment report

```shell
OpenLane Container (a35b64a):/openlane$ python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-69-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11331258/issue.tar.gz)

### Relevant log output

```shell
OpenLane Container (a35b64a):/openlane$ ./flow.tcl -design spm -tag RUN_2023.04.26_09.09.28 -gui
OpenLane a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/spm/config.json'...
[INFO]: PDK Root: /home/matt/work/asic-workshop/shuttle9/pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/spm/runs/RUN_2023.04.26_09.09.28
[WARNING]: A run for spm with tag 'RUN_2023.04.26_09.09.28' already exists. Pass the -overwrite option to overwrite it.
[INFO]: Sourcing /openlane/designs/spm/runs/RUN_2023.04.26_09.09.28/config.tcl. Note that any changes to the DESIGN config file will NOT be applied.
[INFO]: Current ODB: /openlane/designs/spm/runs/RUN_2023.04.26_09.09.28/results/routing/spm.odb
[INFO]: Use 'set_odb file_name.odb' if you'd like to change it.
[ERROR]: during executing openroad script /openlane/scripts/openroad/gui.tcl
[ERROR]: Log: ../dev/null
[ERROR]: Last 10 lines:
13# 0x00007FCC0C12A3F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007FCC1422F3B5 in /lib64/libQt5DBus.so.5
19# 0x00007FCC1054BE71 in /lib64/libQt5Core.so.5
20# 0x00007FCC13CB0EA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
```

mattvenn, 2023-04-26:
This is the same problem as commented here: https://github.com/The-OpenROAD-Project/OpenLane/issues/1149#issuecomment-1271248983

I tried:

`DOCKER_OPTIONS += --privileged`

and now -gui works.

donn, 2023-04-30:
The issue survey is to be run from outside the container, not inside, as the warning states. Please update your issue, as this is highly host-OS specific and we can't do much with the current info.

mattvenn, 2023-05-02:
oops, sorry:

```att-desktop:2032 [(no branch)]: python3 ./env.py issue-survey
Kernel: Linux v5.15.0-71-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

vijayank88, 2023-05-02:
@mattvenn 
Is any specific reason using still two months old tag `2023.02.23` as the latest one is `[2023.04.28](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2023.04.28)`?

mattvenn, 2023-05-02:
Yes, I need a version of the tools that match with shuttle submissions.
This was the tag I used for tiny tapeout on the latest chip ignite, so
that's the one I have installed.