The-OpenROAD-Project/OpenLane #1800: Mismtach between ports for `sky130_fd_sc_hd__tapvpwrvgnd`

kareefardi, 2023-05-11:
### Description

<will add later>

### Expected Behavior

<will add later>

### Environment report

```shell
<will add later>
```

### Reproduction material

<will add later>

### Relevant log output

```shell
<will add later>
```

d-mitch-bailey, 2023-05-11:
@kareefardi I haven't been seeing a problem in device level lvs with this cell abstracted. Maybe there's a problem with lef lvs?

kareefardi, 2023-05-11:
@d-mitch-bailey this mistmatch wouldn't really cause any errors. It is just flooding Verilator with warnings about disconnected pins.

d-mitch-bailey, 2023-05-12:
Is there an option in verilator to ignore certain cells? Don't `fill` cells give the same error.

kareefardi, 2023-12-24:
This was mainly caused by including gate level netlists through `VERILOG_FILES_BLACKBOX`. Since we blackbox these files now, such warnings are not generated. However the main issues still remains.

> Is there an option in verilator to ignore certain cells? Don't `fill` cells give the same error.

Unfortunately, I don't know a way to really do that in Verilator.