The-OpenROAD-Project/OpenLane #1806: Trouble finding documentation about Custom-Building PDKs

Sysynth, 2023-05-16:
After using the sky130 PDK, I was trying to test the tool with a proprietary PDK for a different technology.

In order to do that, the instructions from the "Porting a PDK" documentation section (https://openlane.readthedocs.io/en/latest/for_developers/pdk_structure.html) have been followed and the config files written.

However, I am having trouble finding documentation on how to proceed with the installation of the PDK, since the only thing that I have found is about sky130 and gf180mcu PDKs.

gurolsaglam, 2024-05-16:
I also used the sky130 PDK and wanted to use another open source PDK for a different technology, however there is not enough documentation to prepare the config files. It would be great if an example PDK with the appropriate files could be added.

donn, 2024-09-12:
There is no "installation" of the PDK. You prepare the PDK and point OpenLane towards it by setting the `PDK_ROOT` and `PDK` environment variables.

We have some slightly better documentation in OpenLane 2

closed 2024-09-12 by donn