The-OpenROAD-Project/OpenLane #1808: Sky130 ff libs missing table templates

kareefardi, 2023-05-21:
### Description

STA warns about missing template. Reading ff libs bombards the logs with this warning
```
table template ccsn_ovrf not found.
table template ccsn_ovrf not found.
table template ccsn_ovrf not found.
table template ccsn_ovrf not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_pnlh not found.
table template ccsn_ovrf not found.
table template ccsn_ovrf not found.
table template ccsn_ovrf not found.
table template ccsn_ovrf not found.
```

### Expected Behavior

No warning

### Environment report

```shell
Kernel: Linux v6.1.20-060120-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.22 (OK)
OpenLane Git Version: 54bfaa3d0e826fe64bdc7e6e1049f3600fc28942
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

54bfaa3d 2023-05-11T21:01:58+03:00 [BOT] Update openroad_app (#1774) - Openlane Bot -  (HEAD -> master, origin/master, origin/HEAD)
587fe72b 2023-05-11T14:51:16+03:00 [BOT] Update PDK (#1712) - Openlane Bot -  ()
5a291005 2023-05-10T22:30:22+03:00 Integrate back OpenROAD STA (#1784) - Kareem Farid -  (tag: 2023.05.11)
---
Git Remotes

donn	https://github.com/donn/openlane.git (fetch)
donn	https://github.com/donn/openlane.git (push)
fork	git@github.com:kareefardi/OpenLane.git (fetch)
fork	git@github.com:kareefardi/OpenLane.git (push)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

N/A

### Relevant log output

```shell
line 619, table template ccsn_ovrf not found.
line 625, table template ccsn_ovrf not found.
line 633, table template ccsn_ovrf not found.
line 639, table template ccsn_ovrf not found.
line 647, table template ccsn_pnlh not found.
line 654, table template ccsn_pnlh not found.
line 661, table template ccsn_pnlh not found.
line 668, table template ccsn_pnlh not found.
line 675, table template ccsn_pnlh not found.
line 682, table template ccsn_pnlh not found.
line 689, table template ccsn_pnlh not found.
line 698, table template ccsn_pnlh not found.
line 705, table template ccsn_pnlh not found.
line 712, table template ccsn_pnlh not found.
line 719, table template ccsn_pnlh not found.
line 726, table template ccsn_pnlh not found.
line 733, table template ccsn_pnlh not found.
line 740, table template ccsn_pnlh not found.
line 789, table template ccsn_ovrf not found.
line 795, table template ccsn_ovrf not found.
line 801, table template ccsn_ovrf not found.
line 807, table template ccsn_ovrf not found.
line 815, table template ccsn_ovrf not found.
```
```

daquintero, 2023-09-07:
Is there a particular resolution for this?

delaram-mz, 2024-01-08:
@kareefardi I'm having the same issue, did you find any solution to fix this?

maliberty, 2024-01-08:
FYI, OR doesn't support CCS currently.  Perhaps this is for use in another tool?

kareefardi, 2024-01-16:
@maliberty These are just coming from the PDK without them targeting a specific tool. However, I do think that the warning by OpenROAD is correct and it should be fixed in the PDK file.

@delaram-mz No fix a the moment. They are just warnings and don't cause errors. However, they are noisy and we can attempt to handle them.

young03600, 2025-01-06:
Hi,

I found that some libs will have ccs table in the ```timing()``` even in the lib file without ccs_noise suffix in the file name, e.g., sky130_fd_sc_hd__ss_n40C_1v60 and sky130_fd_sc_hd__ff_n40C_1v95.
This may be the root cause?

Thanks.