The-OpenROAD-Project/OpenLane #1811: OL quits on Verilator error, but they are only warnings

hpretl, 2023-05-22:
### Description

Running the stock `spm` example with OL `2023.05.19` the flow quits on error:

```
/foss/designs > flow.tcl -design spm
OpenLane af89aba6b7fd4c629a97fce12c26c0c8d219a947
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'spm/config.json'...
[INFO]: PDK Root: /foss/pdks
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /foss/designs/spm/runs/RUN_2023.05.22_15.31.55
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running Verilator (log: spm/runs/RUN_2023.05.22_15.31.55/logs/synthesis/verilator.log)...
[ERROR]: 2 errors found by Verilator
[ERROR]: Step 0 (verilator_lint_check) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL run_verilator CALL run_verilator_step CALL {run_non_interactive_mode -design spm}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_verilator" line 35)
    invoked from within
"run_verilator"
    (procedure "run_verilator_step" line 3)
    invoked from within
"run_verilator_step"} -errorline 1
[INFO]: Saving current set of views in 'spm/runs/RUN_2023.05.22_15.31.55/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'spm/runs/RUN_2023.05.22_15.31.55/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'spm/runs/RUN_2023.05.22_15.31.55/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

However, when running Verilator on the spm.v, only two warnings are thrown:

```
/foss/designs > iic-vlint.sh -v spm/src/spm.v 
[INFO] Run verilator linting on spm/src/spm.v...
%Warning-DECLFILENAME: spm/src/spm.v:35:8: Filename 'spm' does not match MODULE name: 'TCMP'
   35 | module TCMP(clk, rst, a, s);
      |        ^~~~
                       ... For warning description see https://verilator.org/warn/DECLFILENAME?v=5.009
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-UNUSEDSIGNAL: spm/src/spm.v:23:20: Signal is not driven, nor used: 'xy'
                                          : ... In instance spm
   23 |     wire[size-1:0] xy;
      |                    ^~
%Error: Exiting due to 2 warning(s)
[DONE] Bye!
```
 @donn Please take a look. Maybe the variable `QUIT_ON_VERILATOR_ERRORS` should be set to 0 for some time being.

### Expected Behavior

OL running `spm` w/o error.

### Environment report

```shell
/foss/tools/openlane/2023.05 > python3 ./env.py issue-survey
Kernel: Linux v5.15.49-linuxkit
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: af89aba6b7fd4c629a97fce12c26c0c8d219a947
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

af89aba6 2023-05-18T13:37:26+03:00 get_tag.py: Don't encode illegal docker tag when local building (#1803) - Darryl Miles -  (HEAD, tag: 2023.05.19)
2ec3bfdd 2023-05-16T21:06:51+03:00 Do not skip STA when `CLOCK_PORT` is not specified (#1797) - Kareem Farid -  (tag: 2023.05.17)
54bfaa3d 2023-05-11T21:01:58+03:00 [BOT] Update openroad_app (#1774) - Openlane Bot -  (tag: 2023.05.12)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

Run stock `spm`

### Relevant log output

```shell
See above.
```

kareefardi, 2023-05-22:
Can you show the log file produced by OpenLane?

```
/foss/designs > iic-vlint.sh -v spm/src/spm.v 
[INFO] Run verilator linting on spm/src/spm.v...
%Warning-DECLFILENAME: spm/src/spm.v:35:8: Filename 'spm' does not match MODULE name: 'TCMP'
   35 | module TCMP(clk, rst, a, s);
      |        ^~~~
                       ... For warning description see https://verilator.org/warn/DECLFILENAME?v=5.009
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-UNUSEDSIGNAL: spm/src/spm.v:23:20: Signal is not driven, nor used: 'xy'
                                          : ... In instance spm
   23 |     wire[size-1:0] xy;
      |                    ^~
%Error: Exiting due to 2 warning(s)
[DONE] Bye!
```
OpenLane uses `--Wno-DECLFILENAME` so this can't be the output produced from OpenLane.

hpretl, 2023-05-22:
[openlane.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11531718/openlane.log)

hpretl, 2023-05-22:
The linting output above is from our own linting script, `iic-vlint.sh`. I posted `openlane.log` above.

kareefardi, 2023-05-22:
I meant verilator log file `spm/runs/RUN_2023.05.22_15.31.55/logs/synthesis/verilator.log`

hpretl, 2023-05-22:
[verilator.log](https://github.com/The-OpenROAD-Project/OpenLane/files/11531760/verilator.log)

Sorry, here the correct one.

hpretl, 2023-05-22:
Ah, I see. You have a WA in your Verilator build script:

```
# workaround for
# %Error: //build/share/verilator/include/verilated_std.sv:28:9: Redeclaring the 'std' package is not allowed
#28 | package std;
#/build/share/verilator/include/verilated_std.sv:2:1: ... note: In file included from verilated_std.sv

RUN rm /build/share/verilator/include/verilated_std.sv &&\
    touch /build/share/verilator/include/verilated_std.sv
```

Does this mean we also need this? Is this bricking something with Verilator? We have just one instance of Verilator for everything, not only for OL.

kareefardi, 2023-05-22:
The interesting part here is that `iic-vlint.sh` didn't produce the same error. I could figure out what triggers the `std` error. If we can understand why this happens with OL, we can remove the WA. I think we can start by comparing `iic-vlint.sh` with the way OL runs it.

hpretl, 2023-05-22:
Here you go: https://github.com/iic-jku/iic-osic/blob/main/iic-vlint.sh

We do just a plan call `verilator --lint-only -Wall "$FILE_NAME"`, without all the lib includes you do. Somewhere from the SKY130 libs the error seems to be triggered.

hpretl, 2023-05-22:
Whenever either this `-I /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v` or `-I /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/primitives.v` is included in the Verilator call it fails with the error.

kareefardi, 2023-05-22:
I just tried. This happens with using any `-I <file>`.

kareefardi, 2023-05-22:
Yes

kareefardi, 2023-05-22:
```
❯ verilator --lint-only -Wall --Wno-DECLFILENAME src/spm.v -I src/spm.v
%Error: //usr/local/share/verilator/include/verilated_std.sv:28:9: Redeclaring the 'std' package is not allowed
   28 | package std;
      |         ^~~
        /usr/local/share/verilator/include/verilated_std.sv:2:1: ... note: In file included from verilated_std.sv
%Error: Exiting due to 1 error(s)
```

hpretl, 2023-05-22:
Ah, know I have a guess. The file share/verilator/include/verilated_std.sv says in the comment, it is automatic included whenever blablabla. What if it is included for **every** file, and of course the second time the `std` package is already declared.

kareefardi, 2023-05-22:
It would be great if you did yes. Thanks.

hpretl, 2023-05-22:
Done, see https://github.com/verilator/verilator/issues/4215

hpretl, 2023-05-23:
@kareefardi The underlying Verilator issues have been fixed (see https://github.com/verilator/verilator/issues/4215), but the OL calling of Verilator needs to be fixed as well.

You call Verilator with

`verilator  --lint-only  -Wall  --Wno-DECLFILENAME  --top-module spm  -I /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v -I /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/primitives.v /foss/designs/spm/src/spm.v -Wno-fatal --relative-includes`

But `-I` shall point to directories, not files! From Verilator help:

`-I<dir>                    Directory to search for includes`

closed 2023-06-19 by kareefardi